sieren/Homepoint #13: Unable to build homepoint on MacOS

rhannink, 2019-06-13:
Hallo,

Thanks for your Great work, I want to complice the code for my M5Stack, but I am unable to complice the code. I installeer all the tools according to the manualen, but when I run:

`cmake -DM5STACK=ON ../`

I het the error:

`-bash: cmake: command not found`

I checken my installation of the tools and all the paths seems to be ok. The file CMakeLists.txt has something to do with it, but it is not being used it seems.

Also running:

`idf.py menuconfig`

Gives the error:

`'cmake' must be available on the PATH to use idf.py`

My .bash_profile looks like:

`export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH
export IDF_PATH=~/esp/esp-idf
export PATH="$IDF_PATH/tools:$PATH"`

Gr. Remco

sieren, 2019-06-13:
Hey,
looks like you don't have `cmake` installed.

The easiest way to go about this on MacOS is to install [Homebrew](https://brew.sh/) and then run `brew install cmake`.

Let me know if that works for you :)

rhannink, 2019-06-13:
Thanks for your fast respons, instelling cake did the trick, But now I run into a new compiling error:

`-- Compiling for M5Stack
CMake Error at main/CMakeLists.txt:118 (spiffs_create_partition_image):
  Unknown CMake command "spiffs_create_partition_image".`

Gr. Remco

mfk-ableton, 2019-06-13:
`Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.` points towards something gone wrong in your PATHs.

Here are mine as a reference (from `zsh.rc`):

```
export IDF_PATH="~/Documents/Development/esp-idf"
export PATH=$PATH:~/Documents/Development/xtensa-esp32-elf/bin
```

I assume you followed the instructions of running
```mkdir build
cd build
cmake -DM5STACK=ON ../
``` ?

sieren, 2019-06-13:
`Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.` points towards something gone wrong in your PATHs.

Here are mine as a reference (from `zsh.rc`):

```
export IDF_PATH="~/Documents/Development/esp-idf"
export PATH=$PATH:~/Documents/Development/xtensa-esp32-elf/bin
```

I assume you followed the instructions of running
```mkdir build
cd build
cmake -DM5STACK=ON ../
```
?

rhannink, 2019-06-13:
Yep,

My .bash_profile contains:

```
export PATH=~/esp/xtensa-esp32-elf/bin:$PATH
export IDF_PATH=~/esp/esp-idf
export PATH="$IDF_PATH/tools:$PATH"
```

And I did
```
cd ~/home/esp/homepoint/build
cmake -DM5STACK=ON ../
```

Gr. Remco

sieren, 2019-06-13:
Does this error appear when you run `make` or `cmake` ?

rhannink, 2019-06-13:
It appears when I run cmake.

When I do:

```MacbookPro:bin remco$ which ld```
I get:

```/usr/bin/ld```

So there is another build environment. Might that be a Problem?

sieren, 2019-06-13:
Yeah, could be. I'd put the `$PATH` variable before the IDF Path in your bash.rc, because OSX will use the first `ld` it can find.

rhannink, 2019-06-13:
Putting $PATh in first makes no different....

sieren, 2019-06-13:
Which ESP-IDF version are you using?

rhannink, 2019-06-13:
3.2

sieren, 2019-06-13:
Ahhh, I see. I just tried that and had the same error as yours.
The issue is that HomePoint relies on CMake which is pretty new with ESP-IDF. I recommend you checkout the `master` of `esp-idf` through github and use that.
That definitely works on my end (although they recently introduced a new warning about using an old GCC version). That situation is not ideal, but thanks for pointing it out.

Let me know if that works on your end?

rhannink, 2019-06-13:
Thanks,

Now its working. Only the last step:

```make -j 8 flash```

Gives a lot of errors, but that is probably because it tries to flash the image to the M5stack ? It is not connectie and I havent deinde a serial port anywhere.

sieren, 2019-06-13:
Try building without `flash` at the end

rhannink, 2019-06-13:
Hmm,

Still no sucess,

```make -j 8 flash and
make -j 8```

Both give a lot (thousands) of these waring below. And 2 errors:

```
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/crypto/sha1.c:135:(.text.sha1_prf+0xa): dangerous relocation: call8: call target out of range: strlen
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/crypto/sha1.c:158:(.text.sha1_prf+0x73): dangerous relocation: call8: call target out of range: memcpy
esp-idf/wpa_supplicant/libwpa_supplicant.a(fast_aes-unwrap.c.obj): In function `fast_aes_unwrap':
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-unwrap.c:37:(.text.fast_aes_unwrap+0xa): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-unwrap.c:39:(.text.fast_aes_unwrap+0x16): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-unwrap.c:58:(.text.fast_aes_unwrap+0x61): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-unwrap.c:60:(.text.fast_aes_unwrap+0x76): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-unwrap.c:65:(.text.fast_aes_unwrap+0x93): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-unwrap.c:66:(.text.fast_aes_unwrap+0x9e): dangerous relocation: call8: call target out of range: memcpy
esp-idf/wpa_supplicant/libwpa_supplicant.a(fast_aes-wrap.c.obj): In function `fast_aes_wrap':
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-wrap.c:42:(.text.fast_aes_wrap+0xa): dangerous relocation: call8: call target out of range: memset
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-wrap.c:43:(.text.fast_aes_wrap+0x16): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-wrap.c:62:(.text.fast_aes_wrap+0x42): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-wrap.c:63:(.text.fast_aes_wrap+0x4d): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-wrap.c:68:(.text.fast_aes_wrap+0x65): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_aes-wrap.c:70:(.text.fast_aes_wrap+0x7b): dangerous relocation: call8: call target out of range: memcpy
esp-idf/wpa_supplicant/libwpa_supplicant.a(fast_sha256-internal.c.obj): In function `fast_sha256_vector':
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256-internal.c:36:(.text.fast_sha256_vector+0xd): dangerous relocation: call8: call target out of range: mbedtls_sha256_starts_ret
esp-idf/wpa_supplicant/libwpa_supplicant.a(fast_sha256.c.obj): In function `fast_hmac_sha256_vector':
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c:112:(.text.fast_hmac_sha256_vector+0x3a): dangerous relocation: call8: call target out of range: memset
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c:112:(.text.fast_hmac_sha256_vector+0x45): dangerous relocation: call8: call target out of range: memcpy
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c:112:(.text.fast_hmac_sha256_vector+0xa2): dangerous relocation: call8: call target out of range: memset
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c:112:(.text.fast_hmac_sha256_vector+0xad): dangerous relocation: call8: call target out of range: memcpy
esp-idf/wpa_supplicant/libwpa_supplicant.a(fast_sha256.c.obj): In function `fast_sha256_prf':
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c:143:(.text.fast_sha256_prf+0x11): dangerous relocation: call8: call target out of range: strlen
/Users/remco/esp/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c:160:(.text.fast_sha256_prf+0x85): dangerous relocation: call8: call target out of range: memcpy
esp-idf/mbedtls/mbedtls/library/libmbedtls.a(esp_hardware.c.obj): In function `mbedtls_hardware_poll':
/Users/remco/esp/esp-idf/components/mbedtls/port/esp_hardware.c:21:(.text.mbedtls_hardware_poll+0x7): dangerous relocation: call8: call target out of range: esp_fill_random
collect2: error: ld returned 1 exit status
make[2]: *** [homepoint.elf] Error 1
make[1]: *** [CMakeFiles/homepoint.elf.dir/all] Error 2
make: *** [all] Error 2
```

sieren, 2019-06-13:
What toolchain version are you using?

sieren, 2019-06-13:
1.22.0-80-g6c4433a-5.2.0 is the version on CI and my computers

rhannink, 2019-06-13:
How can I check?

According to the online Manual I downloaded:

[(https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-macos.tar.gz)]

sieren, 2019-06-13:
Correct, the 8.2.0 is still in the `unstable` manual and did not work last time I checked, the correct one is specified here:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/macos-setup.html

rhannink, 2019-06-13:
Did a new Install of the toolchain version 5.2.0 but Problem remains.....

sieren, 2019-06-13:
I still believe there is something off in the toolchain, cause none of the errors mentions Homepoint specifically, but they appear while compiling the ESP-IDF SDK.

sieren, 2019-06-13:
Running `xtensa-esp32-elf-gcc -v` on command line will tell you which version is in use

rhannink, 2019-06-13:
```gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) ```

sieren, 2019-06-13:
I assume you deleted the build directory and reconfigured?

rhannink, 2019-06-13:
What do you mean by reconfigured?

sieren, 2019-06-13:
Running cmake again

rhannink, 2019-06-13:
Yeah,
After cleaning and erebuilding it works:

```[100%] Built target __idf_build_target
[100%] Built target homepoint.elf
[100%] Built target gen_project_binary
[100%] Built target app
Scanning dependencies of target flash
Note: esptool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: Using default baud rate 460800. To modify, set ESPBAUD environment variable.
esptool.py --chip esp32 -b 460800 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x10000 homepoint.bin 0x1000 bootloader/bootloader.bin 0x310000 spiffs.bin
esptool.py v2.7-dev
Found 3 serial ports
Serial port /dev/cu.SOC```

Sorry for all your effort, my mistake, I lack Some real software engineering skills ;-)

Maybe you can add the part of the masterbranch of **esp-idf** and the correct version off the toolchain to your **README.md**

sieren, 2019-06-13:
Awesome! Glad you got it working 👏

And yeah, I already had that in mind while we were discussing this :D

closed 2019-06-13 by sieren