sieren/Homepoint #15: Project does not compile with current esp-idf master

roamingthings, 2019-06-15:
Hi,

I'm trying to build the project with the current esp-idf master https://github.com/espressif/esp-idf/commit/58df1d93bc17c74499d58e05390af9c309192a5c

The build fails with the follwing error

```
In file included from .../Homepoint/components/arduino-esp32/cores/esp32/wiring_private.h:31,
                 from .../Homepoint/components/arduino-esp32/cores/esp32/wiring_pulse.c:18:
.../Homepoint/components/arduino-esp32/cores/esp32/Arduino.h:121:6: error: conflicting types for 'random'
 long random(long, long);
      ^~~~~~
In file included from .../Homepoint/components/arduino-esp32/cores/esp32/Arduino.h:28,
                 from .../Homepoint/components/arduino-esp32/cores/esp32/wiring_private.h:31,
                 from .../Homepoint/components/arduino-esp32/cores/esp32/wiring_pulse.c:18:
.../xtensa-esp32-elf-8.2.0/xtensa-esp32-elf/sys-include/stdlib.h:267:6: note: previous declaration of 'random' was here
 long random (void);
      ^~~~~~
```

I know it's the (unstable) compiler version 8.2.0. when I use the compiler version 5.2.0 cmake will refuse to build.

I think it's a temporary problem since I already build Homepoint successfully.

I wonder if it's good to relate on the master branch of ESP-IDF. It seems to be pretty unstable. Unfortunatly earlier versions won't support cmake very well.

roamingthings, 2019-06-15:
I just realized that 5.2.0 can be used. It just gives a warning. This SDK is really confusing ;)

closed 2019-06-15 by roamingthings

sieren, 2019-06-15:
Ah glad you got it working, yeah.. It's #1 on my list to update documentation with more details after having gone through https://github.com/sieren/Homepoint/issues/13 ;)

sieren, 2019-06-15:
Regarding cmake: That's unfortunately true, but I think they have a stable cut ready in the not so far future and I'll probably switch Homepoint to that version then :)

roamingthings, 2019-06-15:
I thin Release 3.3 will be the first official cmake release.

kkwestt, 2019-07-04:
also i have error: 

$ cmake -DBUTTONS=ON ../
-- Project version: a75ebb3-dirty
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /Users/1/esp/esp-idf/requirements.txt are satisfied.
CMake Error at /Users/1/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):
  /Users/1/esp/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  /Users/1/esp/esp-idf/tools/cmake/build.cmake:407 (__build_expand_requirements)
  /Users/1/esp/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/1/homepoint/build/CMakeFiles/CMakeOutput.log".
See also "/Users/1/homepoint/build/CMakeFiles/CMakeError.log".

sieren, 2019-07-04:
Which ESP-IDF Version are you using? And which Toolchain version?

kkwestt, 2019-07-04:
$ git describe --tags --dirty
v4.0-dev-1136-g28f1cdf5e

kkwestt, 2019-07-04:
ok
I switched the version with the command
```
cd $IDF_PATH
git fetch
git checkout release/v3.1
git pull
git submodule update --init --recursive
```

sieren, 2019-07-04:
I'm investigating this right now. It looks like the error originates from the ESD-IDF master branch. Version 3.3 is lacking SPIFFS support in Cmake unfortunately, so youre likely to run into another error there. I'm trying to find a commit that works.

kkwestt, 2019-07-04:
yes right now a get new error:
Unknown CMake command "spiffs_create_partition_image".

sieren, 2019-07-04:
Easiest way for now is to use commit `02f2e63662d0314b689bb4925f5d8efa08cfe033` (git checkout 02f2e63662d0314b689bb4925f5d8efa08cfe033) on ESP-IDF while I investigate a long-term solution based on a stable tag.

kkwestt, 2019-07-04:
it's so hard for my brain ))
can you made .bin file ready for upload?

sieren, 2019-07-04:
Unfortunately the JSON is part of the bin file and you want to configure it before you compile it. I'll look into an easier way at some point. 
For now using that commit is the most straight forward approach until i come up with a simpler solution.

kkwestt, 2019-07-04:
maybe port this project to platformio?