sieren/Homepoint #22: Build issue (probably PEBCAK)

rdehuyss, 2019-08-25:
Hi,

I'm trying to build this project but unfortunately encounter a build issue. I think I have a correct environment (including the correct commit version of ESP-IDF).

This is the output I have:
```
rdehuyss@ronald-desktop:~/projects/Personal/m5stack/homepoint/build$ cmake -DM5STACK=ON ../
CMake Warning at /home/rdehuyss/.esp/esp-idf/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/coap/libcoap is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/idf.cmake:42 (__build_init)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/rdehuyss/.esp/esp-idf/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/esp_wifi/lib_esp32 is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/idf.cmake:42 (__build_init)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/rdehuyss/.esp/esp-idf/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/esptool_py/esptool is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/idf.cmake:42 (__build_init)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/rdehuyss/.esp/esp-idf/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/lwip/lwip is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/idf.cmake:42 (__build_init)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/rdehuyss/.esp/esp-idf/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/mbedtls/mbedtls is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/idf.cmake:42 (__build_init)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/rdehuyss/.esp/esp-idf/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/mqtt/esp-mqtt is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/idf.cmake:42 (__build_init)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- ccache will be used for faster builds
-- Project version: c09760c
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /home/rdehuyss/.esp/esp-idf/requirements.txt are satisfied.
CMake Error at /home/rdehuyss/.esp/esp-idf/components/esp32/project_include.cmake:11 (message):
  Internal error, toolchain has not been set correctly by project (or an
  invalid CMakeCache.txt file has been generated somehow)
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:284 (include)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:449 (__build_process_project_includes)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:315 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/home/rdehuyss/projects/Personal/m5stack/homepoint/build/CMakeFiles/CMakeOutput.log".
```

My system:
```
rdehuyss@ronald-desktop:~/projects/Personal/m5stack/homepoint/build$ printenv PATH
/home/rdehuyss/.esp/esp-idf/tools:/home/rdehuyss/.esp/xtensa-esp32-elf/bin:/home/rdehuyss/.local/bin:/home/rdehuyss/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
rdehuyss@ronald-desktop:~/projects/Personal/m5stack/homepoint/build$ printenv IDF_PATH
/home/rdehuyss/.esp/esp-idf
rdehuyss@ronald-desktop:~/projects/Personal/m5stack/homepoint/build$ cd $IDF_PATH
rdehuyss@ronald-desktop:~/.esp/esp-idf$ git status
HEAD detached at 02f2e6366
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   components/coap/libcoap (new commits)
	modified:   components/esp_wifi/lib_esp32 (new commits)
	modified:   components/esptool_py/esptool (new commits)
	modified:   components/lwip/lwip (new commits)
	modified:   components/mbedtls/mbedtls (new commits)
	modified:   components/mqtt/esp-mqtt (new commits)

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	components/bt/controller/
	components/bt/host/

```

Any idea what might be wrong?

sieren, 2019-08-25:
Hey, I think there's a hint in there:
Try going to the directory for esp-idf and run `git submodule update --init --recursive`, could be that you forgot after checking out that specific commit? At least thats what happens to me occasionally :)

rdehuyss, 2019-08-25:
Wauw, fastest reply ever! Thanks for that...

Tried your suggestion, same error:

```
rdehuyss@ronald-desktop:~/projects/Personal/m5stack/homepoint/build$ cmake -DM5STACK=ON ../
-- ccache will be used for faster builds
-- Project version: c09760c
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /home/rdehuyss/.esp/esp-idf/requirements.txt are satisfied.
CMake Error at /home/rdehuyss/.esp/esp-idf/components/esp32/project_include.cmake:11 (message):
  Internal error, toolchain has not been set correctly by project (or an
  invalid CMakeCache.txt file has been generated somehow)
Call Stack (most recent call first):
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:284 (include)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/build.cmake:449 (__build_process_project_includes)
  /home/rdehuyss/.esp/esp-idf/tools/cmake/project.cmake:315 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/home/rdehuyss/projects/Personal/m5stack/homepoint/build/CMakeFiles/CMakeOutput.log".

```

I don't see anything unusual in `CMakeOutput.log` (but I must confess that my make/C/C++ skills are very rusty)

sieren, 2019-08-25:
Are you using the new toolchain for ESP-IDF 4.0+ ? 

If youre on linux, this might help: https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

rdehuyss, 2019-08-25:
That's the one I used...

On this page, they also mention the 5.2 version : https://docs.espressif.com/projects/esp-idf/en/stable/get-started-cmake/linux-setup.html

 :confused:

sieren, 2019-08-26:
Weird, you can take a look at https://github.com/sieren/Homepoint/blob/master/.travis.yml and see if you did the same during setup.

sieren, 2019-08-26:
What I just realised looking at your pastes, you didnt clean ESP-IDF when you switched branches/commits. Always run `git clean -x -f -f -d` and make sure there are no untracked leftovers (in your case, some components from a prior ESP-IDF commit).

I just merged a new PR that updates HomePoint for the `releases/v4.0` branch of ESP-IDF instead of an arbitrary commit.

closed 2019-09-19 by sieren

MostEmpire, 2023-03-28:
I fixed this problem by manually cleaning the project folder and adding -G "CodeBlocks - MinGW Makefiles" generator argument to cmake when making the project, since the implicit generator was visual studio one and it basically wanted to build an uncompilable  cmake garbage out of the project. idf.py uses ninja generator. I did not try that one.

Ramzil11, 2024-04-20:
@MostEmpire Thank you. Your answer helped me in solving this problem