sieren/Homepoint #30: Icon

muaaz1, 2019-09-19:
Hi there, may i know how u add the icons ? Like living room, bedroom icon? 
im still newbie in uploading images into m5stack so can you help me?

sieren, 2019-09-19:
Take a look at the `data` subfolder and make sure to name your images accordingly to the others. Then you can just use then as an `icon` inside the `config.json`.

Images and configuration get copied into the filesystem as part of the `make spiffs_spiffs_bin` build command. Make sure to flash after :)

muaaz1, 2019-09-19:
thank you for your help. 
btw, i have already found the filesystem folder, but may i know how to flash it? is it by running the file from command prompt?

sieren, 2019-09-19:
You simply run `make spiffs_spiffs_bin` before `make flash`. 
make spiffs_spiffs_bin generates the .bin file which represents a separate partition which will be flashed during `make flash`. If you want to learn more about SPIFFS and flashing I recommend the ESP-IDF documentation :)

muaaz1, 2019-09-19:
is the ESP-IDF documentation is this link?
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spiffs.html

muaaz1, 2019-09-19:
the **make spiffs_spiffs_bin** is located in folder? or i have to install any package first?

sieren, 2019-09-19:
There are some detailed build instructions in the README file of this repo :) If you follow those you can run `make spiffs_spiffs_bin`... spiffs_spiffs_bin is just a command for the makefile.

muaaz1, 2019-09-19:
oh okay2 thank you very much for your help sieren. Im really new to this. Will try my best

sieren, 2019-09-19:
No worries. Try googling for some `cmake` and `makefile` tutorials to maybe get a better understanding of how a build-system works. I think that might help :) I'll close this for now.

closed 2019-09-19 by sieren