sieren/Homepoint #37: Error building Homepoint (wifi)

KaVo194, 2019-12-04:
After investing a lot of time to get esp-idf working I'm now encountering an error when building Homepoint for M5Stack with the latest version of esp-idf (master branch):

[make git:(master) ✗] make -j 8 flash
.....
"In file included from /Users/xxx/esp/homepoint/main/wifi/WifiContext.cpp:13:
/Users/xxx/esp/homepoint/main/wifi/WifiContext.cpp: In function 'void event_handler(void*, esp_event_base_t, int32_t, void*)':
/Users/xxx/esp/homepoint/main/wifi/WifiContext.cpp:50:31: error: cannot convert 'esp_ip4_addr_t*' {aka 'esp_ip4_addr*'} to 'const ip4_addr_t*' {aka 'const ip4_addr*'}
                  ip4addr_ntoa(&event->ip_info.ip));
                               ^~~~~~~~~~~~~~~~~~
 ....
make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/wifi/WifiContext.cpp.obj] Error 1
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
make: *** [flash] Error 2"

sieren, 2019-12-04:
Oh no! Maybe the recent master has broken it again... I’ll look into it.
I generally dont recommend `master`. Instead in the esp-idf directory, do `git checkout release/v4.0`. Can you report back if that solves it?

KaVo194, 2019-12-04:
Yes, it resolved the problem.

closed 2019-12-10 by sieren