sieren/Homepoint #38: Installation

fatfox511, 2019-12-10:
Hello Sieren,

thanks for your great work!

Is there any easier way to get homepoint up and running?

Maybe by just flashing a .bin file or an .ino file to the ESP32?

I got crazy with installing the esp-idf development toolkit ;)

Thanks mate :)

sieren, 2019-12-10:
It’s on my list of things to do, but its a bit of work.
ESP32 needs to be put in AP mode, you’d need to connect to it with your phone/computer, set it to the WIFI and then provide a webinterface to upload/change the config.

I’ll look into it when I have some more time, but it’s quite a bit of work.

fatfox511, 2019-12-10:
Sounds like setting up Tasmota :) it would be great.

I look forward to it!

Regards from Potsdam :)

InEc2018, 2019-12-14:
Can´t build the project...

Stops with cmake - get error message : " cl : Command line error D8021: invalid numeric argument '/Wno-frame-address' [C:\Users\mainuser\Desktop\esp-idf\homepoint\build\CMakeFiles\CMakeTmp\cmTC_4cb5c.vcxproj]"

any ideas?

sieren, 2019-12-14:
Are you using the esp toolchain? Something seems off, cause I see a visual studio project file in your error message

InEc2018, 2019-12-14:
Thanks for the very quick reply!

I installed "esp-idf-tools-setup-2.1.exe" which includes the toolchain.

But i have also installed visual studio 2019 on the same pc, maybe that will cause the problem... 

i will try to remove the path-vars from visual studio temporarly

InEc2018, 2019-12-14:
Tested it on a new VM:

My Steps:
Installed 
esp-idf-tools-setup-2.1.exe, downloaded git / python and installed ESP-IDF release 4.0
extracted the "esp32_win32_msys2_environment_and_toolchain-20181001" to Drive C:

Tried to get and build the project with the following steps from your Guide:
$ git clone https://www.github.com/sieren/homepoint.git
$ cd homepoint
$ git submodule update --init --recursive
$ mkdir build && cd build
$ cmake ../
$ make spiffs_spiffs_bin
$ make -j 8 flash

faild on step "cmake ../" with the following message `:C:\Users\Admin\Desktop\esp-idf\homepoint\build>cmake ../
-- Building for: NMake Makefiles
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Admin/.expressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Admin/.expressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
CMake Error: Generator: execution of make failed. Make command was: "nmake" "/nologo" "cmTC_b46ac\fast"
-- Check for working C compiler: C:/Users/Admin/.expressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- broken
CMake Error at C:/Users/Admin/.expressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "C:/Users/Admin/.expressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/Admin/Desktop/esp-idf/homepoint/build/CMakeFiles/CMakeTmp

    Run Build Command:"nmake" "/nologo" "cmTC_b46ac\fast"
    Das System kann die angegebene Datei nicht finden
    Generator: execution of make failed. Make command was: "nmake" "/nologo" "cmTC_b46ac\fast"

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/Admin/Desktop/esp-idf/tools/cmake/project.cmake:258 (__project)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Admin/Desktop/esp-idf/homepoint/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Admin/Desktop/esp-idf/homepoint/build/CMakeFiles/CMakeError.log".`

sieren, 2019-12-14:
Looks like your CMake wants to use NMake makefiles and I guess Nmake cant be found. I haven't developed in Win machines in a while so I'm a bit rusty, but you can try passing "-G Ninja" to CMake when configuring as this is the recommended build tool by Espressif and should've been installed by default with their tools:

https://docs.espressif.com/projects/esp-idf/en/stable/get-started-cmake/windows-setup.html

Leviathan09, 2020-01-18:
> It’s on my list of things to do, but its a bit of work.
> ESP32 needs to be put in AP mode, you’d need to connect to it with your phone/computer, set it to the WIFI and then provide a webinterface to upload/change the config.

With Arduino IDE you can select a HTML Website base OTA sample which is running on a ESP32
You just need to enter the Wifi credentials
On the Website then you can upload the .bin file
So uploading a .bin file isn't that complicated i think.

Is there a "easy" way to put your proejct into a .bin file to flash it to the ESP32?

Like "InEc2018" i'm new to all of this.
So all above is just a suggestion
As i don't know if putting Homepoint into a .bin file is the "bit of work" or the part with the upload to a ESP32.

Would be nice to hear from you
Hopefully you find the time to maybe work on this

sieren, 2020-01-18:
Hey, the part of uploading is the bit of work. I was looking into this over the christmas holidays, but the easy solution with the Arduino SDK is currently not working since the ESP32 Arduino Library hasnt been updated for 4.0 yet. I suppose they'll do that in the coming months which is when I'll try and make all of this work :)

sieren, 2020-02-09:
Happy to report I'm working on an uploadable .bin ... gonna come in a few days

fatfox511, 2020-02-09:
> Happy to report I'm working on an uploadable .bin ... gonna come in a few days

This is the best thing i have read in 2020 :)

Thank you!

sieren, 2020-02-12:
What devices is everyone planning to run this on?
I can provide binaries for M5Stack and for generic ESP32 devices that follow the PIN Layout of ArduiTouch

sieren, 2020-02-12:
There's also a PR about to be merged introducing tons of new features for this (like initial AP mode after first flash, failsafe mode when configuration parsing fails, error message displays etc, see https://github.com/sieren/Homepoint/pull/47 )

Leviathan09, 2020-02-12:
> 
> 
> What devices is everyone planning to run this on?
> I can provide binaries for M5Stack and for generic ESP32 devices that follow the PIN Layout of ArduiTouch
I'm planning to use a NodeMCU ESP32 ([this one](http://anleitung.joy-it.net/?goods=nodemcu-esp32))
Don't know if this is a generic one, but i think so

cbrightly, 2020-02-12:
> What devices is everyone planning to run this on?
> I can provide binaries for M5Stack and for generic ESP32 devices that follow the PIN Layout of ArduiTouch

I have several versions of the M5Stack I am hoping to get this uploaded to.

sieren, 2020-02-13:
That's great!
Does anyone mind testing this?
[homepoint_beta1.zip](https://github.com/sieren/Homepoint/files/4196955/homepoint_beta1.zip)

All you need is esptool from here:
https://github.com/espressif/esptool/releases

Install with `python setup.py install` and then run
`esptool.py write_flash -fs 4MB -fm dout 0x0 homepoint_m5stack.bin` (or the other .bin for a generic ESP32/ArduiTouch version)

Happy to hear about feedback!

sieren, 2020-02-13:
It's done!
I've published a release with instructions, added the front page with all information and did a short video walking through the initial setup process: https://www.youtube.com/watch?v=Ql9BQKl_Yuk

closed 2020-02-13 by sieren