sieren/Homepoint #39: last Step fail

Caspar311, 2019-12-23:
Hi Matt, 
a big thank for your great work.

I spend the last two Days to install Homepoint on my ESP32. After changing from Windows to an Linux PC it was a little bit easier. :)
But actually now i cant go further.

I cant find the mistake.

Thanks a lot for helping.

<pre> 85%] <font color="#4E9A06">Building C object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-time.c.obj</font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-touch.c:18</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/ets_sys.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-touch.c:20</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/esp_intr.h:16:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-sigmadelta.c:19</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/ets_sys.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-misc.c:34</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/rtc.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:18</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/ets_sys.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:20</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/esp_intr.h:16:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:21</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/gpio.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/gpio.h is deprecated, please use esp32/rom/gpio.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/gpio.h is deprecated, please use esp32/rom/gpio.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-spi.c:20</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/ets_sys.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-spi.c:22</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/esp_intr.h:16:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-spi.c:23</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/gpio.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/gpio.h is deprecated, please use esp32/rom/gpio.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/gpio.h is deprecated, please use esp32/rom/gpio.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-timer.c:19</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/rom/ets_sys.h:1:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
<b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:</b> In function &apos;<b>__pinMode</b>&apos;:
<b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:90:24:</b> <font color="#EF2929"><b>error: </b></font>&apos;<b>rtc_gpio_desc</b>&apos; undeclared (first use in this function); did you mean &apos;<b>rtc_io_desc</b>&apos;?
     uint32_t rtc_reg = <font color="#EF2929"><b>rtc_gpio_desc</b></font>[pin].reg;
                        <font color="#EF2929"><b>^~~~~~~~~~~~~</b></font>
                        <font color="#4E9A06">rtc_io_desc</font>
<b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:90:24:</b> <font color="#34E2E2"><b>note: </b></font>each undeclared identifier is reported only once for each function it appears in
In file included from <b>/home/florian/esp/homepoint/components/arduino-esp32/cores/esp32/esp32-hal-timer.c:23</b>:
<b>/home/florian/esp/esp-idf/components/esp32/include/esp_intr.h:16:2:</b> <font color="#AD7FA8"><b>warning: </b></font>#warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [<font color="#AD7FA8"><b>-Wcpp</b></font>]
 #<font color="#AD7FA8"><b>warning</b></font> esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  <font color="#AD7FA8"><b>^~~~~~~</b></font>
esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/build.make:206: recipe for target &apos;esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-gpio.c.obj&apos; failed
make[3]: *** [esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-gpio.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:4925: recipe for target &apos;esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/all&apos; failed
make[2]: *** [esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/all] Error 2
CMakeFiles/Makefile2:214: recipe for target &apos;CMakeFiles/flash.dir/rule&apos; failed
make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
Makefile:216: recipe for target &apos;flash&apos; failed
make: *** [flash] Error 2
</pre>

sieren, 2019-12-23:
Hey,

which ESP-IDF version are you using? Most of the times these build errors come from peopel trying to use ESP-IDF 3.3 or ESP-IDF Master... I recommend either grabbig the 4.0 beta release or checking out the `releases/v4.0` branch on the esp-idf repo.

Caspar311, 2019-12-23:
Hey, thank you for the fast answer.
Your right, i use ESP-IDF 3.3
I'll try the 4.0 beta version first.
Thank you very much.

Caspar311, 2019-12-28:
Hello Matt,
after three full days of testing i decided to wait until there was an easier installation option. With Linux everything worked out, but then I couldn't persuade the laptop to connect my ESP32 and under Windows the C / C ++ Compiler could not be found at the end or not in the PATH.
Do you happen to plan to do it through PlatformIO?
Best wishes
Florian

sieren, 2020-01-03:
I might look into this in the future, once ESP IDF 4.0 is PlatformIO ready

closed 2020-01-03 by sieren