sieren/Homepoint #43: Upload error after compilation finished on OSX

tschamm, 2020-01-03:
Hi,

I have a problem installing the homepoint.elf to my ESP32 at the end of the compilation:

```
[ 98%] Linking CXX static library libmain.a
[100%] Built target __idf_main
[100%] Generating ld/esp32.project.ld
[100%] Built target __ldgen_output_esp32.project.ld
[100%] Linking CXX executable homepoint.elf
[100%] Built target homepoint.elf
[100%] Generating binary image from built executable
esptool.py v2.8
Generated ~/Homepoint/build/homepoint.bin
[100%] Built target gen_project_binary
[100%] Built target app
Note: esptool.py will attempt to set baud rate automatically. To specify a baud rate, set the ESPBAUD environment variable.
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'spiffs.bin'
esptool.py --chip esp32 -p /dev/tty.SLAB_USBtoUART write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 homepoint.bin 0x310000 spiffs.bin
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

make[3]: *** [CMakeFiles/flash] Error 1
make[2]: *** [CMakeFiles/flash.dir/all] Error 2
make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
make: *** [flash] Error 2
```
Any idea what's going wrong? I use esp-idf release/4.0, but I don't know where to get this spiffs.bin from.

Regards, Thomas

sieren, 2020-01-03:
Hey Thomas,
It’s in the README under Build Instructions.
My guess is you forgot to run `make spiffs_spiffs_bin` before running the make flash command.

Don’t forget you need to create a new SPIFFS Image every time you modify the JSON file, as this is where the filesystem lives.

Hope that helps :)

Matt

tschamm, 2020-01-04:
Thanks a lot, Matt. I run this command once, but forgot to run it again later.

closed 2020-01-04 by tschamm