sieren/Homepoint #45: Trouble compiling on a Pi

mk1blacklimited, 2020-01-27:
I've (finally) managed to install ESP-IDF successfully, but now the last make fails. Any tips?

```
make -j 8 flash
[  0%] Built target esp32_linker_script
[  0%] Built target _project_elf_src
[  1%] Built target partition_table
[  1%] Performing build step for 'bootloader'
[  1%] Built target __idf_app_trace
[  2%] Built target __idf_cxx
[  1%] Built target _project_elf_src
[  4%] Built target __idf_log
[  4%] Built target __idf_newlib
[ 12%] Built target __idf_xtensa
[  4%] Built target __idf_vfs
[ 50%] Built target __idf_soc
[  6%] Built target __idf_freertos
[ 53%] Built target __idf_micro-ecc
[  6%] Built target __idf_heap
[ 79%] Built target __idf_bootloader_support
[  6%] Built target __idf_log
[ 87%] Built target __idf_efuse
[  9%] Built target __idf_soc
[ 90%] Built target __idf_spi_flash
[ 10%] Built target __idf_esp_rom
[ 93%] Built target __idf_main
[ 11%] Built target __idf_esp_common
[ 98%] Built target bootloader.elf
[ 12%] Built target __idf_xtensa
[100%] Built target gen_project_binary
[100%] Built target app
[ 14%] Built target __idf_esp32
[ 14%] No install step for 'bootloader'
[ 14%] Built target __idf_espcoredump
[ 14%] Completed 'bootloader'
[ 14%] Built target __idf_pthread
[ 15%] Built target bootloader
[ 16%] Built target __idf_esp_event
[ 16%] Built target __idf_tcpip_adapter
[ 24%] Built target __idf_lwip
[ 25%] Built target __idf_esp_eth
[ 26%] Built target __idf_esp_wifi
[ 27%] Built target __idf_nvs_flash
[ 28%] Built target __idf_spi_flash
[ 28%] Built target __idf_app_update
[ 29%] Built target __idf_bootloader_support
[ 30%] Built target __idf_efuse
[ 38%] Built target __idf_wpa_supplicant
[ 40%] Built target mbedtls
[ 47%] Built target mbedcrypto
[ 48%] Built target mbedx509
[ 50%] Built target __idf_driver
[ 50%] Built target __idf_esp_ringbuf
[ 50%] Built target __idf_console
[ 51%] Built target __idf_spiffs
[ 52%] Built target __idf_sdmmc
[ 53%] Built target __idf_openssl
[ 54%] Built target __idf_wear_levelling
[ 56%] Built target __idf_nghttp
[ 57%] Built target __idf_asio
[ 57%] Built target __idf_esp_adc_cal
[ 59%] Built target __idf_coap
[ 59%] Built target __idf_esp_gdbstub
[ 59%] Built target __idf_protobuf-c
[ 60%] Built target __idf_expat
[ 60%] Built target __idf_jsmn
[ 61%] Built target __idf_json
[ 61%] Built target __idf_ulp
[ 62%] Built target __idf_unity
[ 65%] Built target __idf_freemodbus
[... 105 lines cut ...]
                 from /home/pi/esp/esp-idf/homepoint/main/ui/UIStatusBarWidget.cpp:1:
/home/pi/esp/esp-idf/homepoint/main/util/warnings.h:26: warning: "RAPIDJSON_ASSERT" redefined
 #define RAPIDJSON_ASSERT(x) if(!(x)) throw std::logic_error("JSON Parsing failed. Check your config!");
 
In file included from /home/pi/esp/esp-idf/homepoint/main/libraries/rapidjson/include/rapidjson/allocators.h:18,
                 from /home/pi/esp/esp-idf/homepoint/main/libraries/rapidjson/include/rapidjson/reader.h:20,
                 from /home/pi/esp/esp-idf/homepoint/main/libraries/rapidjson/include/rapidjson/document.h:20,
                 from /home/pi/esp/esp-idf/homepoint/main/mqtt/MQTTSensorUtils.hpp:5,
                 from /home/pi/esp/esp-idf/homepoint/main/mqtt/MQTTSensorGroup.hpp:4,
                 from /home/pi/esp/esp-idf/homepoint/main/model/Model.hpp:5,
                 from /home/pi/esp/esp-idf/homepoint/main/AppContext.h:3,
                 from /home/pi/esp/esp-idf/homepoint/main/ui/UIStatusBarWidget.h:5,
                 from /home/pi/esp/esp-idf/homepoint/main/ui/UIStatusBarWidget.cpp:1:
/home/pi/esp/esp-idf/homepoint/main/libraries/rapidjson/include/rapidjson/rapidjson.h:406: note: this is the location of the previous definition
 #define RAPIDJSON_ASSERT(x) assert(x)
 
/home/pi/esp/esp-idf/homepoint/main/ui/UIButton.cpp: In member function 'virtual void gfx::UIButton::draw()':
/home/pi/esp/esp-idf/homepoint/main/ui/UIButton.cpp:56:20: warning: variable 'centerPoint' set but not used [-Wunused-but-set-variable]
         const auto centerPoint = mFrame.getCenterPoint();
                    ^~~~~~~~~~~
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:167: esp-idf/main/CMakeFiles/__idf_main.dir/ui/UIWidget.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
```

sieren, 2020-01-27:
The only error-message I see is `fatal error: Killed signal terminated program cc1plus` which doesn't give away much.
My suspicion is that the Pi runs out of memory.
Can you run `dmesg` on the pi immediately after and see if there's something along the lines of `Out of memory: Kill process 23747 (cc1plus) score 15 or sacrifice child` ?

mk1blacklimited, 2020-01-27:
You might be right there - trying a larger swap file now...

mk1blacklimited, 2020-01-27:
Jeez, sorry...:) It worked, thanks a lot! 👍

sieren, 2020-01-28:
Glad to see it worked :)

closed 2020-01-28 by sieren