sieren/Homepoint #60: Battery level

Dosh22, 2020-02-27:
Hi Sieren, can you optimize homepoint to battery operation (sleep of ESP32, poweroff screen in 1min) and add battery level monitoring and show % in status bar... Standard 3.7-4.2 Lipo battery 2500mah, sense via analog in A0 (with voltage divider of course)

sieren, 2020-03-29:
Hey, yeah, I can probably take a look at that. I assume you're using the button version? Not sure how to wake an ESP32 from touch-screen.

radokristof, 2020-04-07:
@sieren Sorry to interrupt, which devices do you use? I know I can build a custom one, but is there any which is a complete HW beside M5Stack? I would need something with touchscreen and USB powered.

sieren, 2020-04-07:
I'd look for the ArduiTouch + AZDelivery NodeMCU ESP32 (https://www.amazon.com/AZDelivery-Nodemcu-CP2102-Module-Development/dp/B07F877YZQ) 

It can be powered by USB (if plugged into the board on the ArduiTouch) or just 9-24V external power.