sieren/Homepoint #64: Potential issue with standard build script

jholthusen, 2020-03-19:
Hi Matt,
just setup an Ubuntu VM to double check my own documentation. When running the scripts as described in the README.md and uploading them directly via the script, I get an usuable ESP32. When using the build_all script, I can flash the resulting file with esptool.py and have a working Homepoint.

Could you double check? This is against HEAD, directly cloning https://www.github.com/sieren/homepoint.git

Could double check this works for you?

Thanks!
Jan

sieren, 2020-03-19:
That's probably because you manually need to swap out DEFINES for generic ESP32 or M5Stack. The build-script does this automatically by swapping the `User_Setup.h` file in the TFT_eSPI Library folder.

If you don't do that you might end up with a black screen.

jholthusen, 2020-03-19:
Hmm. looking at the User_setup.h, it seems to default to ArduiTouch... which is my hardware setup. In addition, I will check, but I believe I even have issues on the serial Monitor, which should have no impact, right?

Jan

sieren, 2020-03-19:
It shouldn't. However if you want more serial output, you should add `Serial.begin(115200);` somewhere in the init functions in `main.cpp`, otherwise it stops after the RTOS Bootup-phase

jholthusen, 2020-03-19:
I checked again. And rest my case. Now it (also) works on my PC. 
Apologies for that...

Jan

closed 2020-03-19 by jholthusen

sieren, 2020-03-19:
No worries :)