sieren/Homepoint #71: [Feature Request] ADD Home-Assistant

thundergreen, 2020-03-25:
Dear Dev. That's the most genial Project for me to use it as a small touch panel to control my AC in each room. Unfortunately I use Home-Assistant. Do you plan to make it also running for Home Assistant?

sieren, 2020-03-29:
I'm afraid not for now. I'm working on Hue integration next, HomeAssistant would mean lots more work and on top of that I'm unfortunately not overly familiar with the platform.

sieren, 2020-04-16:
Did you consider using the MQTT Module in Home Assistant to talk to HomePoint?

thundergreen, 2020-04-16:
Yes I do :)

blackout314, 2020-10-09:
> 
> 
> Dear Dev. That's the most genial Project for me to use it as a small touch panel to control my AC in each room. Unfortunately I use Home-Assistant. Do you plan to make it also running for Home Assistant?

and, it works?

cbrightly, 2020-10-09:
@thundergreen What doesn’t currently work with Home Assistant? If you have an MQTT integration, it can be set up to have the Homepoint device discovered automatically, correct?

sieren, 2021-01-16:
Closed, discussion moved to https://github.com/sieren/Homepoint/issues/130

closed 2021-01-16 by sieren