sieren/Homepoint #72: ESP32 + ILI9341 touch reversed

konzolec, 2020-03-27:
I'm using generic bin on ESP32 with SPI 240x320 ILI9341 (Touch) but my touch commands are not responding correctly. It seems they're reversed. 

Video can be found here: https://vimeo.com/user106054719/review/401289240/1bbdfed996

Any idea what could cause this?

sieren, 2020-03-27:
I'd love to help, can you give me more details? (Like what brand is the display?)
The video doesn't work either unfortunately.

konzolec, 2020-03-27:
Sorry wrong link. 
https://vimeo.com/401289240

This is the display im using:
#Aliexpress € 6,54  11%OFF | 3.2 inch 320*240 SPI Serial TFT LCD Module Display Screen with Touch Panel Driver IC ILI9341 for MCU
https://a.aliexpress.com/_d8fiH8e

sieren, 2020-03-29:
That's odd... I recomplied the generic bin with a different rotation angle, do you mind trying it with this one?

[homepoint_espgeneric.bin.zip](https://github.com/sieren/Homepoint/files/4398648/homepoint_espgeneric.bin.zip)

konzolec, 2020-03-29:
Touch is working now correctly! :) Thanks

sieren, 2020-03-29:
Interesting.. not exactly sure why this is happening, maybe something with inverted coordinates on different IL9341 screens. I might have to expose this in the config file and make it configurable...

closed 2020-04-05 by sieren