sieren/Homepoint #73: Screensaver after inactivity

konzolec, 2020-04-04:
After inactivity i get a white screen, when i press it i get the Homepoint menu back. Is this on purpose?

Using ESP32 and ILI9341 LCD https://a.aliexpress.com/_d8fiH8ew

[https://vimeo.com/404035313](url)

sieren, 2020-04-04:
Guess everything is a bit upside down with the unit you got yourself there. 
In all the pictures and videos I have seen from others so far, I haven't seen any of these issues.

The expected behaviour is: Black screen, no backlight. 
The Screensaver basically does 2 things: Disables the TFT_LED (GPIO15) and also writes a separate command to the screen to switch off the entire power to the screen.

(See `main/config/Config.h` L62)

konzolec, 2020-04-04:
Thanks for clarifying that. I have TFT_LED connected to +3.3V because i have a black screen while connected to GPIO15. There is zero voltage. Back to the drawing board and figuring out why is that.

sieren, 2020-04-04:
That's a bit against the point of open source. I can't really provide custom images for people in the long run, that's why the code is all here.
But I will make a mental note to add this to the JSON file (configurable screensaver duration incl. switching it off).

In the meantime, here are the schematics for which the generic version is built:
https://www.hwhardsoft.de/app/download/11468539297/ArduiTouch+ESP+schematic+V01-03-00.pdf
See the screen on the top right, the NodeMCU ESP32 unit on the top left. Maybe this helps figuring out why your display unit is behaving so weirdly.

sieren, 2020-04-04:
Let me know if this one does the trick

[homepoint_espgeneric.bin 2.zip](https://github.com/sieren/Homepoint/files/4432390/homepoint_espgeneric.bin.2.zip)

konzolec, 2020-04-04:
I can confirm after the datasheet you provided the generic and invertion fix build work! I had to add the the extra resistor and BC557 transistor. Thank you again for your help!

closed 2020-04-04 by konzolec

sieren, 2020-04-04:
So you had to apply both? Add the transistor but also use the custom build with the inversion fix? 
I think I'll add the invertion stuff to the config too then. shouldnt be too hard.

konzolec, 2020-04-05:
Yes, I used the custom invertion fix and transistor part for screensaver. Now the screen goes black as it supposed to.

sieren, 2020-04-05:
Ok cool!

I have a PR open https://github.com/sieren/Homepoint/pull/74 so with the next official release, these should be configurable through the config.json file and won't need any custom builds.