sieren/Homepoint #77: Add beep on event.

llevet, 2020-04-07:
Hi,

Add possibility to make a beep on a event (getTopic and onValue or offValue)  : 

For example : 
{
      "name": "Porte 1",
      "setTopic": "dumy/garage/porte/status/set",
      "getTopic": "dumy/garage/porte/status",
      "onValue": "open",
      "offValue": "close
      "alarmValue": "open,5"
	},

In this example, it generate a beep during 5 seconds when getTopic take the value "open"
This alarm is ringing only with external event (ie i don't have push button on ESP to change the state)

Thanks.

llevet, 2020-04-07:
And same for sensor : 

  {
    "name": "Bedroom",
    "type": "Sensor",
    "icon": "door",
    "devices": [{
      "name": "Bedroom Temp",
      "type": "singleValue",
      "firstIcon": "temperature_small",
      "jsondata": false,
      "firstKey": "Temp",
      "getTopic": "/WEMOS-BED1/BME280/Temp"
      "alarmValue": "30,5"
    },

In this example, it generate a beep during 5 seconds when getTopic take the value egual of upper to 30

Thanks.

sieren, 2020-04-07:
Before I'd take this on, can you describe me your use-case from a users point of view?
Just trying to figure out how much actual value there is in me taking on this work.

Otherwise you can always add this yourself, since all of this is open source :)

llevet, 2020-04-07:
For sensor, i use ESPEasy (https://github.com/letscontrolit/ESPEasy) with GAZ sensor and Mosquitto publication. I want to have a alarm if there is a certain high value.

Same for Garage door is opening when i haven't give the order to open by the touchscreen of the arduitouch.

AZ-Touch ESP (arduitouch from Zihatec) is perfect interface console for that.

llevet, 2020-04-08:
Also, when a alarm is ringing, screen must be wake up (i think especially to alarm on sensor, because sensor don't wake up the screen ).