sieren/Homepoint #78: Bug on device on main screen

llevet, 2020-04-07:
Device activation on main screen change the status immediately without get the return of value getTopic.
It doesn't appear on device secondary screen (devices in Device-group are OK). 

Thanks.

llevet, 2020-04-21:
Any news ? 
This is problematic to see the status changing on click even if the device is disconnected or does not work properly.

Thanks.

llevet, 2020-04-21:
Sadly my C++ competence is too bad !
Thanks to your Radar :-)

sieren, 2021-02-14:
TODO: Remove setting the Group State after tapping it - instead rely on getTopic. This was intended to be a feature and address the shortcomings of ESP32's slow performance, but seems to be more confusing to people based on this post.