sieren/Homepoint #80: Error : Detected overlap at address: 0x310000 for file: spiffs.bin

MLindek, 2020-04-09:
Hi Matt,
i just setup an Debian 10_3 VM to play with an ESP32 and Homepoint. 
When running the scripts as described in the README i get an Error. 
When using the build_all script I get the same Error.

With "make -j 8 flash" gives the Error: 
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x310000 for file: spiffs.bin
esptool.py --chip esp32 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 homepoint.bin 0x310000 spiffs.bin

and with the build_all script i get the Error: 
Exception: Not possible to create this bin, overlapping between spiffs.bin and homepoint.bin

Can you help me ?

sieren, 2020-04-09:
Interesting, can you give me some more details? Mostly: Which ESP-IDF version are you using and which ESP32 board?

sieren, 2020-04-09:
Nevermind, i'm seeing the same issue

sieren, 2020-04-09:
Thanks for the report... just updated master with a fix. Guess this got introduced with the recent changes.

closed 2020-04-09 by sieren