sieren/Homepoint #89: Tasmota Read Status seams not to work

franki29, 2020-04-18:
Hi, sorry to bother again. I add a switch for a Tasmota SON/OFF  .
Switching On/Off works, but if I switch the device with a other device like the Tasmota device webpage, Homepoint does not show the actual device status.
"scenes": [{
    "name": "Strompumpe",
    "type": "Switch",
    "icon": "garden",
    "devices": [{
      "name": "Strompumpe",
      "setTopic": "Strompumpe/cmnd/power1",
      "getTopic": "Strompumpe/stat/power1",
      "onValue": "ON",
      "offValue": "OFF"

I also have a device with Espurna Firmware and with that it works fine.

Br. Frank

sieren, 2020-04-18:
Hey Frank,

to further diagnose this, I suggest connecting to the MQTT Broker with something like https://chrome.google.com/webstore/detail/mqttlens/hemojaaeigabkbcookmlgmdigohjobjm?hl=en 
and observing any broadcasts when using the webpage.

alternatively `mosquitto_sub -v -h <mqtt_ip_address> -p <port> -t '#'`.

I just tried it myself and I could see broadcasts coming from Tasmota in the form of:
`stat/gosund3/POWER ON`

Let me know about your findings.

M

franki29, 2020-04-18:
Hi, my device is A RGB Plug with RGB LED ring, that can be also switch on/off. Therfore  I have Power1 & Power2. Here my MQTT message.
Client mosqsub|4136-raspberryp received PUBLISH (d0, q0, r0, m0, 'Strompumpe/stat/POWER1', ... (3 bytes))
Strompumpe/stat/POWER1 OFF

sieren, 2020-04-18:
Just to be on the safe side, can you use all-caps for `power1` in your configuration?
`"getTopic": "Strompumpe/stat/power1",` -> `"getTopic": "Strompumpe/stat/POWER1",`

franki29, 2020-04-18:
Hi, yes that was the problem. Tasmota seems not to care about caps letters for switch the Plug On/Off. This is the reason I did not look about the right incoming message.
Thanks for helping

Frank

sieren, 2020-04-18:
Thanks for reporting though!
I'll see if I can fix this... it's not the first time this has been a problem.

sieren, 2020-04-18:
Nevermind, just read up on the official specification and topics are supposed to be case sensitive.

closed 2020-04-18 by sieren