sieren/Homepoint #91: Display ILI9341 not working

MC-1984, 2020-04-20:
Hi Matt

I am really new to playing with ESP32´s so it is likely that there is no bug in your code and it is just me not getting it right. 

I have an NodeMCU ESP32 and an ILI9341. Both are working fine when I use TFT_eSLI library with Arduino IDE. I did not get it to work with the PIN configuration shown in the video so I tried it with the exact same configuration I used with the Arduino IDE. That is not working either.

I edited the User_Setup.h file in esp-idf/homepoint/main/config/TFT-eSPI/Generic. I hope that is correct?

I would really appreciate it if you could help!

Markus

`
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
// bi-directional SDA pin and the library will try to read this via the MOSI line.
// To use the SDA line for reading data from the TFT uncomment the following line:

// #define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 display only

// For ST7789 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display

//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

// For M5Stack ESP32 module with integrated ILI9341 display ONLY, remove // in line below

// #define M5STACK

// For ST7789, ST7735 and ILI9163 ONLY, define the pixel width and height in portrait orientation
// #define TFT_WIDTH  80
// #define TFT_WIDTH  128
// #define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
// #define TFT_HEIGHT 160
// #define TFT_HEIGHT 128
// #define TFT_HEIGHT 240 // ST7789 240 x 240
// #define TFT_HEIGHT 320 // ST7789 240 x 320

// For ST7735 ONLY, define the type of display, originally this was based on the
// colour of the tab on the screen protector film but this is not always true, so try
[... 219 lines cut ...]
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS
`

sieren, 2020-04-20:
Hey Markus,

Try esp-idf/homepoint/main/libraries/tft_espi/user_setup.h

This file gets overwritten with esp-idf/homepoint/main/config/TFT-eSPI/Generic.h every time CMake is run, so if you want it persistent, you can copy it back there once you figured out a PIN configuration that works for you :)

The pre-compiled version (generic) uses the following wiring: https://sae5d500d7e085f42.jimcontent.com/download/version/0/module/11468539297/name/ArduiTouch%20ESP%20schematic%20V01-03-00.pdf (NodeMCU on top left)

Not sure if I got any pins wrong in the readme.txt, I think I double-checked at the time

MC-1984, 2020-04-20:
Hey Matt

Thanks for your fast reply. I tried your PIN configuration and the PIN configuration that works on another device. Do you have any idea what I could do wrong? I can find the Wifi and connect. It is just the display that its not working...

Markus

MC-1984, 2020-04-20:
I checked it. You did not get any pins wrong in the readme.txt. But with this configuration the display does not even light up. With the pin configuration I use that works with the Arduino IDE and TFT-eSPI the VCC pin at the display is also connected. That pin is not connected to anything now. And what keeps me wondering is that nothing is connected to 3.3 V (which is connected twice in the other configuration).

sieren, 2020-04-20:
Even in the schematics link i posted above VCC is connected to 3.3V (see the top left).
Guess I need to add that to the readme, I was mostly focussing on the variable software part there expecting most people would wire up VCC/GND themselves.

MC-1984, 2020-04-20:
Thanks again. I already tried connection the VCC to 3.3V. If it is connected while the ESP32 reboots the onboard LED starts flashing fast (does that signal an error?). Otherwise nothing more happens. Do I have to wire anything else not mentioned in the readme? In my other configuration 3.3V is also connected to LED (which is connected to pin15 as your readme says).

Sorry for all the questions. I feel like an idiot for for not getting this simple setup to work...

sieren, 2020-04-21:
When connected to PIN15, try changing https://github.com/sieren/Homepoint/blob/265c0cd8c9ac5bebd145aa832fa0276e685f6cb3/main/config/Config.h#L52 to `true`

MC-1984, 2020-04-21:
Thanks again. Unfortunately, it is still not working. 

When I connect the ESP after flashing to the Arduino IDE I get the following error that repeats about every 350ms

`10:50:17.760 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
10:50:17.760 -> flash read err, 1000
10:50:17.760 -> ets_main.c 371 
10:50:18.074 -> ets Jun  8 2016 00:22:57
`

The second I disconnect 3.3V it boots:

`
10:50:18.111 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:50:18.111 -> configsip: 0, SPIWP:0xee
10:50:18.111 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:50:18.111 -> mode:DIO, clock div:1
10:50:18.111 -> load:0x3fff0018,len:4
10:50:18.111 -> load:0x3fff001c,len:6720
10:50:18.111 -> ho 0 tail 12 room 4
10:50:18.111 -> load:0x40078000,len:12612
10:50:18.111 -> load:0x40080400,len:3788
10:50:18.111 -> entry 0x40080684
10:50:18.111 -> [0;32mI (67) boot: Chip Revision: 1[0m
10:50:18.111 -> [0;32mI (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
10:50:18.152 -> [0;32mI (42) boot: ESP-IDF v4.0-dirty 2nd stage bootloader[0m
10:50:18.152 -> [0;32mI (42) boot: compile time 21:12:42[0m
10:50:18.152 -> [0;32mI (42) boot: Enabling RNG early entropy source...[0m
10:50:18.152 -> [0;32mI (47) boot: SPI Speed      : 80MHz[0m
10:50:18.152 -> [0;32mI (51) boot: SPI Mode       : DIO[0m
10:50:18.152 -> [0;32mI (55) boot: SPI Flash Size : 4MB[0m
10:50:18.152 -> [0;32mI (59) boot: Partition Table:[0m
10:50:18.152 -> [0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
10:50:18.177 -> [0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
10:50:18.177 -> [0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
10:50:18.177 -> [0;32mI (85) boot:  2 factory          factory app      00 00 00010000 00350000[0m
10:50:18.177 -> [0;32mI (92) boot:  3 spiffs           Unknown data     01 82 00360000 0007d000[0m
10:50:18.215 -> [0;32mI (100) boot: End of partition table[0m
10:50:18.215 -> [0;32mI (104) boot_comm: chip revision: 1, min. application chip revision: 0[0m
10:50:18.215 -> [0;32mI (111) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2503f0 (2425840) map[0m
10:50:19.010 -> [0;32mI (906) esp_image: segment 1: paddr=0x00260418 vaddr=0x3ffb0000 size=0x033a0 ( 13216) load[0m
10:50:19.010 -> [0;32mI (911) esp_image: segment 2: paddr=0x002637c0 vaddr=0x40080000 size=0x00400 (  1024) load[0m
10:50:19.010 -> [0;32mI (913) esp_image: segment 3: paddr=0x00263bc8 vaddr=0x40080400 size=0x0c448 ( 50248) load[0m
10:50:19.045 -> [0;32mI (941) esp_image: segment 4: paddr=0x00270018 vaddr=0x400d0018 size=0xd9cc0 (892096) map[0m
10:50:19.328 -> [0;32mI (1230) esp_image: segment 5: paddr=0x00349ce0 vaddr=0x4008c848 size=0x07c2c ( 31788) load[0m
10:50:19.328 -> [0;32mI (1254) boot: Loaded app from partition at offset 0x10000[0m
10:50:19.364 -> [0;32mI (1255) boot: Disabling RNG early entropy source...[0m
10:50:19.364 -> [0;32mI (1255) cpu_start: Pro cpu up.[0m
10:50:19.364 -> [0;32mI (1259) cpu_start: Application information:[0m
10:50:19.364 -> [0;32mI (1264) cpu_start: Project name:     homepoint[0m
10:50:19.364 -> [0;32mI (1269) cpu_start: App version:      v0.02-3-g52da495-dirty[0m
10:50:19.364 -> [0;32mI (1275) cpu_start: Compile time:     Apr 10 2020 21:12:53[0m
10:50:19.398 -> [0;32mI (1282) cpu_start: ELF file SHA256:  b14ed47e9165d559...[0m
10:50:19.398 -> [0;32mI (1288) cpu_start: ESP-IDF:          v4.0-dirty[0m
10:50:19.398 -> [0;32mI (1293) cpu_start: Starting app cpu, entry point is 0x40081288[0m
10:50:19.398 -> [0;32mI (0) cpu_start: App cpu up.[0m
10:50:19.398 -> [0;32mI (1303) heap_init: Initializing. RAM available for dynamic allocation:[0m
10:50:19.398 -> [0;32mI (1310) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
10:50:19.435 -> [0;32mI (1317) heap_init: At 3FFBC2B8 len 00023D48 (143 KiB): DRAM[0m
10:50:19.435 -> [0;32mI (1323) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
10:50:19.435 -> [0;32mI (1329) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
10:50:19.435 -> [0;32mI (1336) heap_init: At 40094474 len 0000BB8C (46 KiB): IRAM[0m
10:50:19.435 -> [0;32mI (1342) cpu_start: Pro cpu start user code[0m
10:50:19.468 -> [0;32mI (1365) spi_flash: detected chip: generic[0m
10:50:19.468 -> [0;32mI (1366) spi_flash: flash io: dio[0m
10:50:19.468 -> [0;32mI (1366) cpu_start: Starting scheduler on PRO CPU.[0m
10:50:19.468 -> [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
10:50:19.468 -> [E][esp32-hal-cpu.c:125] removeApbChangeCallback(): not found func=400F4670 arg=3FFB1398
10:50:19.788 -> [E][vfs_api.cpp:22] open(): File system is not mounted
10:50:19.822 -> [E][vfs_api.cpp:22] open(): File system is not mounted
10:50:19.890 -> [0;31mE (2201) example: Failed to find Font[0m
10:50:19.890 -> I (2213) wifi: wifi driver task: 3ffd179c, prio:23, stack:3584, core=0
10:50:19.890 -> I (2224) wifi: wifi firmware version: 581f422
10:50:19.890 -> I (2224) wifi: config NVS flash: enabled
10:50:19.890 -> I (2224) wifi: config nano formating: disabled
10:50:19.928 -> I (2225) wifi: Init dynamic tx buffer num: 32
10:50:19.928 -> I (2229) wifi: Init data frame dynamic rx buffer num: 32
10:50:19.928 -> I (2234) wifi: Init management frame dynamic rx buffer num: 32
10:50:19.928 -> I (2240) wifi: Init management short buffer num: 32
[... cut ...]
10:50:19.928 -> I (2253) wifi: Init dynamic rx buffer num: 32
10:50:20.032 -> [0;32mI (2344) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0[0m
10:50:20.032 -> I (2345) wifi: mode : softAP (24:6f:28:b6:20:49)
10:50:20.032 -> I (2346) wifi: Total power save buffer number: 16
10:50:20.032 -> I (2349) wifi: Init max length of beacon: 752/752
10:50:20.032 -> I (2354) wifi: Init max length of beacon: 752/752
`

I tried two different USB cables and 2 different power sources. So I don´t think it´s a power issue.

sieren, 2020-04-21:
When you say "flashing to the Arduino IDE", what does that mean?

The first error looks like a hardware issue to me, where the boot sequence is unable to read flash storage.

one thing missing from your configuration is `TFT_LED`, it should be the same PIN number as `TFT_BL`

MC-1984, 2020-04-21:
What I mean is after flashing it via the terminal with 

`esptool.py write_flash -fs 4MB -fm dout 0x0 homepoint_espgeneric.bin`

I open the Arduino IDE and look what messages come in.

I copied the User_Setup.h file from the sketch where the display works perfectly and used the pin configuration from that sketch. But that did not work either. Is there any other configuration file that might not to be changed?

sieren, 2020-04-21:
What commands are you running to get that .binary file and before flashing?

The `Cmake` command that's being run by the build_all script copies & overwrites User_Setup.h in the TFT_ESPI folder in main/libraries/tft_espi (with the one from https://github.com/sieren/Homepoint/blob/master/main/config/TFT_eSPI/Generic/User_Setup.h )

So if you're running the build script each time, all changes you'd have made to the User_Setup file in main/libraries/tft_espi  would be overwritten before compilation.

MC-1984, 2020-04-21:
To be sure I changed the files in both folders.

I am not running any commands before

`esptool.py write_flash -fs 4MB -fm dout 0x0 homepoint_espgeneric.bin`

I thought that command would use the changed user_setup.h file when flashing?

sieren, 2020-04-21:
Oh, no ...that just flashes the same firmware over and over again to the ESP32.
If you'd like the changes to be incorporated, you'd have to compile the full program as described in the build section of the README.

MC-1984, 2020-04-21:
That explains a lot...

Unfortunately, I can only copy and paste the code. I do not understand what each line of code means. Could you please tell me which lines I have to copy and paste to compile the program and flash the ESP? 

I assume that is when the cmake command overwrites the user-setup.h in main/libraries/tft_espi?

I really appreciate you helping me so much!

sieren, 2020-04-21:
I'm afraid there's not much more I can do other than point you to the firmware.
The ESP-IDF document is a good resource on how to setup the toolchain on your computer to actually compile all of this:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/

A much easier alternative would be to give the wiring from the README another shot (and the schematic above) - especially since you already own a NodeMCU ESP32 -  and try using the pre-compiled version.

MC-1984, 2020-04-21:
It´s not a problem to use the wiring from the README. But I still have to do some changes in the user_setup.h file. It is configured for the ArduiTouch and not for the ESP32 board. So I still have to get the changed user_setug.h file flashed onto the ESP, right?

sieren, 2020-04-21:
ArduiTouch is based on NodeMCU ESP32, it is just a hardware board a NodeMCU ESP32 attaches to [1]. Whether you actually have that board or not does not matter. It's down to the wiring.

User_Setup.h is a header file to be used at compile time not at runtime. Therefore it will only be used if you compile the code yourself (which you are not when you use esptool to flash the pre-compiled binary)

[1] https://www.hwhardsoft.de/english/projects/arduitouch-esp/

MC-1984, 2020-04-21:
Okay, but as I apparently never uploaded the modified user_setup.h file to the ESP it should work with the wiring from the readme (plus GND to GND and VCC to 3.3V). Am I correct? Unfortunately, it does not work. I double checked the wiring. Is there anything else I could have done wrong? I can connect to HomePoint-Config. That works so the flashing must have worked.

sieren, 2020-04-21:
try adding

```
{
....,
"ledPinPullup": true,
....
}
```

to the config.json ...if that doesnt work, try with false. 
Could be related to https://github.com/sieren/Homepoint/issues/73
specifically https://github.com/sieren/Homepoint/issues/73#issuecomment-609064729

MC-1984, 2020-04-21:
It´s working!

But it is really strange. When the display is connected to 3.3V I cannot flash the ESP 

`serial.serialutil.SerialTimoutException: Write timeout`

It is working with "ledPinPullup": false, true and without the line in the config.json.

The only thing I had to do was to also connect pin 15 to 3.3V

Thank you so much Matt!

closed 2020-04-21 by MC-1984

sieren, 2020-04-21:
Interesting, I haven't heard of that before. But thanks for reporting!

Glad you got it working :)