sieren/Homepoint #92: Plans to productionize Project builds?

cbrightly, 2020-04-20:
I have taken advantage of the flashing from pre-built app straight to firmware, but the build is stuck at the first release. Are there any plans to update this programmatically or by hand?

sieren, 2020-04-21:
Can you elaborate what you mean? OTA Updates?

cbrightly, 2020-04-22:
To clarify, I was referring to updating the pre-built firmware you released to being kept up to date with the master branch.

sieren, 2020-04-22:
I'm confused, I updated the build here:
https://github.com/sieren/Homepoint/releases/tag/v0.03

Homepoint takes up too much space for an OTA Update right now, so you'd have to reflash it manually for the time being :/

cbrightly, 2020-04-22:
Manual flashing is okay! I missed the new release, thanks for pointing me that way. I didn’t know if you had considered integrating those release builds so they happen automatically on a push to master.

sieren, 2020-04-22:
I thought about doing the full CD approach, but quite often its just not worth it because the fixes and additions are rather incremental so far (plus I'd like to test master locally before rolling it out to everyone, since an update isnt as easy as on desktop or mobile)

cbrightly, 2020-04-22:
> I thought about doing the full CD approach, but quite often its just not worth it because the fixes and additions are rather incremental so far (plus I'd like to test master locally before rolling it out to everyone, since an update isnt as easy as on desktop or mobile)

devel or alpha branches? It would make it much more seamless to the end-user to always have a current build available; the toolchain requirements and very specific and have (at least personally) been a huge hurdle, which is part of why I prefer the releases and their prebuilt images.

I think the ease of flashing firmware vs. building from source code is a barrier to entry for at least some users and that it may be considering some roadmap or a schedule for releasing updates builds. Just a thought!

sieren, 2020-04-22:
Yeah, I feel like a lot of people feel the same way.

But right now I'm still the sole maintainer of this project and there hasn't been a big feature or improvement in a while. Mostly just incremental improvements based on user feedback.

Of course, if I feel like an addition was critical to operation or is a massive improvement I'd do a new release rather sooner than later.

cbrightly, 2020-04-22:
Good to know!
Unrelated, except as maybe a possible feature for inclusion in the future, but have you considered adding Nextion HMI displays to the list of supported hardware? I feel like Homepoint would work naturally with something like this… Home Assistant Switch Plate or HASP comes to mind.

sieren, 2020-04-22:
Technically if it's supported by TFT_eSPI (https://github.com/Bodmer/TFT_eSPI) it should work pretty much out of the box.
I'm considering adding support for Raspberry Pi Zero next but that night be a big undertaking on its own.

sieren, 2020-05-03:
Working on OTA Updates next, which hopefully alleviates issues for people who don't want to connect and flash each single time:
https://github.com/sieren/Homepoint/pull/97

cbrightly, 2020-05-03:
I’m looking forward to trying OTA updates! For whatever reason the toolchain requirements hate me, so not needing to build from source while still enjoying the added features and fixes of new releases is great.

cbrightly, 2020-09-21:
It appears GitHub has solved this for us!

https://github.com/apps/release-drafter

sieren, 2020-09-21:
Nice, I'll use this next time I'm doing a new release. Thanks for the pointers :)

closed 2020-09-21 by sieren