sieren/Homepoint #99: Possible to add a GPIO IN for move detection ?

franki29, 2020-05-11:
Hi,
as the display is switching off after some time (of course a very good function), is it possible to add a move detection PIN to switch it ON? Idea is that instead pressing the touch screen, a simple sensor for PIR or another sensor that detect movement (door is open..) will activate the screen. Nice if this PIN can be configured via the config file and be able to configure if the detection it is at a high or low signal.

Thanks, and best regards
Frank

toehrling, 2020-06-02:
Hi,
i allready builded this and also added some small things to get it in the software running. I replaced the Screensaver functionality and replaced it with the PIR sensor functionality. Probbably there is a way for a factory implementation to this. For me it was the first time with this programming language since ages so my skills are not the best ;). The most PIR sensors allready debounce the movements and have a adjustable time to drop the detection. The screensaver functionality is not nessesary any more in this case. It could get improoved by using an interupt and let the ESP32 sleep and wakeup.

I also added a temperature and humidity sensor to my case. The software part for this is a little more complicated and I stuck a little with problems, because the screen and processor produce heat if active and that make the results a little unprecise. Hopefully this can be solved with another position of the sensor.

BR Thomas

in Screenaver.hpp i done something like this:
`      {
        mLastTouch = std::chrono::system_clock::now();
        pinMode(TFT_LED, OUTPUT);
        pinMode(PIR_MOTION_SENSOR, INPUT);
        digitalWrite(TFT_LED, LOW);    // LOW to turn backlight on - pcb version 01-02-00
      }
`
...
`     template<typename T>
      bool tapped(const T& tapEvt)
      {
        if (!tapEvt)
        if (!tapEvt && !digitalRead(PIR_MOTION_SENSOR))
        {
          return mCurrentState;
        }
        auto tapEvent = *tapEvt;
        if (tapEvent.state == decltype(tapEvent.state)::Tap)
        {
          mLastTouch = std::chrono::system_clock::now();
        }
        if (digitalRead(PIR_MOTION_SENSOR)) {
          mLastTouch = std::chrono::system_clock::now();
        }
        return mCurrentState;
      }`

really crud implementation

toehrling, 2020-07-01:
see here: https://www.thingiverse.com/thing:4513823 I also added humidity and temperature to the thing.

Thias66, 2020-11-19:
Hello, I added a AM312 PIR Sensor (GPIO36). Is there any chance to get some hints for an integration? I would like to enable the screen and transmit the detected motion over mgtt.
And Matt, nice work! Thanks!

toehrling, 2020-11-19:
Hello, just define PIR_MOTION_SENSOR with 36 or the integer that is defined for 36 and add the code of my previews comment to the ScreenSaver.hpp. For the MQTT messages you will find code examples in other files of the project. It not make sense to commit my actual code. I need to merge all the commits to the project that ar betweeen and the stuff is not running on an actual version of esp builder.

Thias66, 2020-11-19:
Ok, at the moment I use a precompiled bin file. I assume I have to compile my own build. Could you tell my witch  file I have to edit? Would I define PIR_MOTION_SENSOR in the ScreenSaver.hpp to? And how would I assign a MQTT topic to the PIR?
Thanks for your support!

Digl, 2022-01-19:
Hi! I would like to be able to control a relay connected to the homepoint esp32.
Do you have any suggestions on how to get that done?
Thanks!