sieren/Homepoint #100: ledPinPullup

Hennoloe, 2020-06-05:
Cant figure out where to set option ledPinPullup

sieren, 2020-06-05:
You need to connect to the open Wifi that gets created (HomePoint-Config) and open http://192.168.99.1 to access the config

Looks like this is missing from the README (it's usually shown on display so I can see where the confusion comes from)

Hennoloe, 2020-06-05:
The ESP Chip is already configured, and is in my Network. Already did the steps you did in the Video, by adding the Text for MQTT and so on.
but my display i think got the ledPinPullup the wrong way, because i cant see anything.

sieren, 2020-06-05:
I see.
Here's an example that seemed to have worked:
https://github.com/sieren/Homepoint/issues/91#issuecomment-617338120

Hennoloe, 2020-06-05:
I already did that, but im not sure what this function does.
on my ESP32 Board the blue LED is blinking right when i touch the Panel, so i think that works.
but i dont get any image.

sieren, 2020-06-05:
If you flashed with the Generic ESP32 Image, please check the wiring instructions here:https://github.com/sieren/Homepoint/wiki/Hardware

(Specifically the PinOut and ArduiTouch Wiring as references)

Hennoloe, 2020-06-09:
I dont know what im doing wrong, the display is white but nothing appears on it.

Hennoloe, 2020-06-09:
I saw an Issue #73 , where someone added the extra resistor and bc557 transistor to fix it.
I think i need this too.
Just wired LED pin to 3,3v, and it works

closed 2020-12-08 by sieren