sieren/Homepoint #101: Problem with the touch screen...

THBSLtd, 2020-07-09:
Hi,

I had to replace the touch screen and now the right and left side of the touch function are reversed. The display and above and below of the touch function still works perfectly. Is there a way to swap right and left side of the touch function?

Many thanks in advance, greetings Christian

sieren, 2020-08-05:
Hey, sorry for the late response. I just merged a fix and created a new build which I've attached to this Issue.
Do you mind checking if this works for you?

I have introduced two new parameters to the config file.
You might wanna try out both of them depending on orientation. Let me know if this fixes it for you.

| Key   | Type | Value                                                                                             |
| --------------| -----| ----------------------------------------------------------------------------------------- |
| "touchXAxisInverted" | Bool | default: `false`, invert the X Axis of the touch screen in case it is misaligned |
| "touchYAxisInverted" | Bool | default: `true`, invert the Y Axis of the touch screen in case it is misaligned |

THBSLtd, 2020-08-05:
Hi,

no Problem :-). I loaded the new bin into the IC and got "Could not parse failsafe file" on the display. When I write the current version in the IC the AP mode opens. Am I doing something wrong? Thank you in advance for your help, 

greetings Christian

sieren, 2020-08-06:
Ahh, sorry, there was a bug in the build script for OTA Releases.
Should be fixed now, attached a new version.

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5034489/homepoint_release.zip)

sieren, 2020-08-06:
Sorry, I realised there were a bunch of bugs in the build script that led to this.
Attached is a new version.

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5035303/homepoint_release.zip)

THBSLtd, 2020-08-07:
Hello, I tested the software and was able to add the new commands to the config file. unfortunately nothing changes with the different settings. I gave the orders as follows 
....
  "timezone": "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00",
  "touchXAxisInverted": false,
  "touchYAxisInverted": true,
  "scenes": [{
....

Is that the right way to do the commands?

Thank you in advance for your help,

greetings Christian

THBSLtd, 2020-08-07:
ps.: I tried diffrent versions of the command as true, true ... true, false and so on..

sieren, 2020-08-07:
hey Christian,

thats odd! Can you give this version a try? I radically swapped the x and y axis in this build.

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5040568/homepoint_release.zip)

THBSLtd, 2020-08-07:
Hello Matt, this is really weird. I've now used a different display and the touch function is different again. Slowly I think that I am writing the command in the wrong place. Can you tell me again where I put it in? Does it go in config.json or somewhere else?

Thank you in advance for your help,

greetings Christian

sieren, 2020-08-07:
Please use the one you've used previously, else it's really difficult for me to pinpoint the issue here.
It does indeed go into config.json

THBSLtd, 2020-08-07:
In the moment it is the same Problem with both displays, no change of the Touch with the diffrent settings :-(.

sieren, 2020-08-07:
Has anything changes with the other version?

sieren, 2020-08-07:
On  a second note: ive tried and verified what you mentioned in the original post (reversal of sides) with the parameters mentioned here on an aduitouch with the il9341 yesterday for this release: https://github.com/sieren/Homepoint/releases/tag/0.04rc1

sieren, 2020-08-07:
Another point.. After applying those changes did you properly reboot the unit or just hit "reload"?

THBSLtd, 2020-08-07:
I reboot it over the Webfrontend, should i take the power away for am moment?

THBSLtd, 2020-08-07:
Hi Matt,

i found my problem... I used all the time the homepoint_espgeneric.bin file and now i used the homepoint_espgeneric_full file and it worked. I used the Version befor the last one, that one from yesterday.

With the Version from today the invertion works the wrong way, But the version from Yesterday works fine. I tested all combinations and it work the right way.

Thanks a lot for your Help, i wish you an nice Weekend,

greetings Christian

sieren, 2020-08-07:
Nice! Thanks for the report. The one you mention is also the official release. Glad we got this sorted 🎉☺️

closed 2020-08-07 by sieren