sieren/Homepoint #109: Icon and German

Obelix411, 2020-08-21:
it is also possible to include your own icons

it is also possible to use German words e.g. Wohnwand 

translated with Google translator

sieren, 2020-08-21:
Ja zu beidem ;)
Die Beschreibungen der Szenen muss man als User sowieso selber angeben.
Die Icons kann man auch austauschen.

Beispiel:
```
{
    "name": "Bedroom",
    "type": "Sensor",
    "icon": "door",
    "devices": [{
      "name": "Temperature DHT Sensor",
      "type": "singleValue",
      "jsondata": true,
      "firstIcon": "temperature_small",
      "firstKey": "temperature",
      "getTopic": "bedroom/esptemp"
    }
```

Hier heißt das icon `door`, Homepoint sucht dann nach `door_active.jpg` und `door_inactive.jpg`.

Obelix411, 2020-08-21:
Vielen Dank für die schnelle Antwort und den deutschen Support bin ein Anfänger und versuche mich ranzulassen

Obelix411, 2020-08-30:
Hi und einen schönen Sonntag ich weis nicht ob du mir helfen kannst ich habe grade einen Temperatursensor angelegt in deiner Software jetzt bekomme ich keine Daten zu ihm hin ich arbeite mit iobroker kannst du mir da helfen?

sieren, 2020-08-30:
Hey, da kenne ich mich leider nicht mit aus. Home point nutzt MQTT als Protokoll.
Häufige Fehlerquelle dort ist dass mqtt topics groß/Kleinschreibung-sensitive sind, sprich auf die groß und kleinschreibung geachtet werden muss.
Dann ist auch zu beachten ob die Temperatur Daten direkt über das topic gehen (zb als float) oder als JSON.

Obelix411, 2020-08-30:
So es hat geklappt habe die zeile mit jsondata raus genommen

Obelix411, 2020-08-30:
Eine frage habe ich noch wie klappt das mit den unter menüs also mit den 3 punkten

closed 2020-08-31 by sieren