sieren/Homepoint #111: m5Stack Basic, seeing a restart about every 60 min

jonathanmbradshaw, 2020-09-10:
M5STACK Basic, 
Seeing a crash about every 60 min.

Project name:     homepoint
App version:      v0.02b-26-gfbdd68e-dirty
Compile time:     Aug 18 2020 17:45:03

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6736
ho 0 tail 12 room 4
load:0x40078000,len:12612
load:0x40080400,len:3788
entry 0x40080684
I (66) boot: Chip Revision: 1
I (70) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0.1-378-g26d7fb774-dirty 2nd stage bootloader
I (41) boot: compile time 17:44:52
I (43) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 80MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00010000 00150000
I (101) boot:  4 ota_1            OTA app          00 11 00160000 00150000
I (108) boot:  5 spiffs           Unknown data     01 82 002b0000 0007d000
I (116) boot: End of partition table
I (120) boot_comm: chip revision: 1, min. application chip revision: 0
I (127) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x483b0 (295856) map
I (232) esp_image: segment 1: paddr=0x000583d8 vaddr=0x3ffb0000 size=0x03968 ( 14696) load
I (238) esp_image: segment 2: paddr=0x0005bd48 vaddr=0x40080000 size=0x00400 (  1024) load
I (239) esp_image: segment 3: paddr=0x0005c150 vaddr=0x40080400 size=0x03ec0 ( 16064) load
I (254) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xd9b94 (891796) map
I (546) esp_image: segment 5: paddr=0x00139bb4 vaddr=0x400842c0 size=0x103c0 ( 66496) load
I (583) boot: Loaded app from partition at offset 0x10000
I (583) boot: Disabling RNG early entropy source...
I (584) cpu_start: Pro cpu up.
I (588) cpu_start: Application information:
I (592) cpu_start: Project name:     homepoint
I (597) cpu_start: App version:      v0.02b-26-gfbdd68e-dirty
I (604) cpu_start: Compile time:     Aug 18 2020 17:45:03
I (610) cpu_start: ELF file SHA256:  1132603f7460be96...
I (616) cpu_start: ESP-IDF:          v4.0.1-378-g26d7fb774-dirty
I (623) cpu_start: Starting app cpu, entry point is 0x400812e8
I (609) cpu_start: App cpu up.
I (633) heap_init: Initializing. RAM available for dynamic allocation:
I (640) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (646) heap_init: At 3FFBCAF8 len 00023508 (141 KiB): DRAM
I (652) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (659) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (665) heap_init: At 40094680 len 0000B980 (46 KiB): IRAM
I (671) cpu_start: Pro cpu start user code

jonathanmbradshaw, 2020-09-14:
is there a 'needs more info' recipe that I should use?
all the best Jonathan

sieren, 2020-09-14:
Hey Jonathan, 
not quite. I'll look into this and will try to reproduce it on my end.
But I also got the idea of adding a "DEBUG" option to the config file so it prints out a stack trace on crash (which it currently doesnt, which is why you only see the start sequence and rst reason on subsequent reboots after crashes)

Thanks for asking though :)

Best
Matt

sieren, 2020-09-14:
Oh, one thing though: can you paste your config here? (dont forget to scrub out passwords and alike)

jonathanmbradshaw, 2020-09-14:
Hi Matt, Thanks for that, sad to say new in this space and still getting my
head around the basics.
I have cut the config file back to just the WiFi, MQTT, and login settings,
and I'll start moving forward from there.
I plan to start adding bits back in and as soon as I see the problem again
I will revert.

Great job on the software, it looks great
more later, all the best Jonatha

sieren, 2020-09-14:
Ah sorry, I meant can you paste your config here (without the login info) so I can take a look in case I might need to reproduce something?

jonathanmbradshaw, 2020-09-15:
Following is the code that was failing, I am working on this to see if I can get the fault using a smaller config file.   all the best Jonathan 
```
{
  "wifi": "*",
  "password": "*",
  "login": "admin",
  "webpass": "*",
  "mqttbroker": "mqtt://*",
  "mqttusername": "",
  "mqttpasswd": "",
  "timezone": "GMT+0IST-1,M3.5.0/01:00:00,M10.5.0/02:00:00",
  
  "scenes": [
    {"name": "Boost",
       "type": "Switch",
       "icon": "power",
       "devices": [
      {"name": "boost",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "power",
      "firstKey": "value",
      "getTopic": "*/boiler_plant/switch/boiler_plantp4_dhw_boost_pump/state",
      "setTopic": "*/house/dhw",
      "onValue": "ON",
      "offValue": "OFF"
}]},
    {"name": "BED",
       "type": "Switch",
       "icon": "power",
       "devices": [
      {"name": "bed",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "power",
      "firstKey": "value",
      "getTopic": "*/libr_manifold_1/switch/relay_2/state",
      "setTopic": "*/libr_manifold_1/relay_2",
      "onValue": "ON",
      "offValue": "OFF"
}]},{"name": "BATH",
       "type": "Switch",
       "icon": "power",
       "devices": [
      {"name": "bath",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "power",
      "firstKey": "value",
      "getTopic": "*/libr_manifold_1/switch/relay_1/state",
      "setTopic": "*/libr_manifold_1/relay_1",
      "onValue": "ON",
      "offValue": "OFF"
}]},{"name": "Fire",
      "type": "Sensor",
      "icon": "fan",
      "devices": [
     {"name": "burner",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "fan",
      "firstKey": "onOff",
      "getTopic": "*/boiler_plant/switch/boiler_plantp3_hw_burner/state"},
     {"name": "dhw",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "fan",
      "firstKey": "onOff",
      "getTopic": "*/boiler_plant/switch/boiler_plantp4_dhw_boost_pump/state"}
 ]},{"name": "H2O",
      "type": "Sensor",
      "icon": "fan",
      "devices": [
     {"name": "tank",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "temperature_small",
      "firstKey": "temp",
      "getTopic": "*/boiler_plant/sensor/tank/state"},
     {"name": "dhw",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "temperature_small",
      "firstKey": "temp",
      "getTopic": "*/boiler_plant/sensor/dhw/state"}
 ]},{"name": "Outside",
      "type": "Sensor",
      "icon": "fan",
      "devices": [
     {"name": "temp",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "temperature_small",
      "firstKey": "temp",
      "getTopic": "*/garden/sensor/wasabi_temp/state"},
     {"name": "hum",
      "type": "singleValue",
      "jsondata": false,
      "firstIcon": "humidity_small",
      "firstKey": "hum",
      "getTopic": "*/garden/sensor/wasabi_humidity/state"}
 ]}]}

sieren, 2021-01-16:
Hey @jonathanmbradshaw - just checking in. Did you upgrade HomePoint to one of the newer versions and if so, are yous till seeing this issue?

jonathanmbradshaw, 2021-01-16:
Just did the upgrade, I'll let you know how it goes.  BTW, great job on the new software, looking good so far.

jonathanmbradshaw, 2021-01-23:
Did the update and its looking great, no issues so far.  I will start
rebuilding my config and I'll let you know if I see any problems.
Nice work, BTW I note that performance around updating the config file has
improved a lot!

closed 2021-02-14 by sieren