sieren/Homepoint #112: M5Stack Grey display inverted

tomnz, 2020-09-17:
Not sure if anyone else is seeing this... As per the title, the display on my M5Stack Grey with the latest Homepoint release (v0.04rc1) seems to be inverted - the background is white, icons are blue instead of yellow, etc. Not a big deal as it's all functional otherwise, but kind of strange. The Grey uses a newer display (ILI9342C), so maybe doesn't work exactly like other M5Stack devices.

I noticed in the Adafruit_ILI9341 library the following function:

https://github.com/adafruit/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.h#L149

Maybe this should be exposed as a configuration option?

Great project, thanks!

sieren, 2020-09-17:
Nice! Thanks for pointing that out. I'll see if I can get a build done for you to test soon since I dont have that specific device.

Cheers

tomnz, 2020-09-17:
Awesome, happy to load and test a precompiled binary! I'm stuck on esp-idf 3.2 for another project at the moment so might be a hassle for me to compile.

sieren, 2020-09-18:
Hey Tom,

do you mind trying this build out?
There's now a `displayColorInverted` option (bool) for the config file. If you set that one to true, it should call the function in the library you mentioned.

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5247062/homepoint_release.zip)

tomnz, 2020-09-18:
Nice, works perfectly thanks!

sieren, 2020-09-18:
Thanks for reporting back :) Just merged this to master

closed 2020-09-18 by sieren

htvekov, 2021-01-04:
Hi' @sieren

FYI
`displayColorInverted: true` also needed for M5stack black basic edition as well.

Struggled a bit with this, until I found that files on release [link](https://github.com/sieren/Homepoint/releases) are from August and not the September files as linked to0 in this issue.

After reprogramming my M5stack with September files, display colors are now correct 😉👍

Ciao !

sieren, 2021-01-05:
@htvekov thanks for the info! I'm getting confused by M5Stacks offerings, since I also have a basic black edition but that one sports the IL9341 Display.
Can you send me a link as to where you've bought yours?

I'll make sure the new release goes up today.