sieren/Homepoint #115: Use of displayColorInverted

mherke, 2020-10-14:
Hi, 

first of all: Thanks for sharing this great Project!

I am running my first tests with a m5stack and so far it looks pretty nice. But right now it is unclear how to use "displayColorInverted". I created a simple test config.json (see below), but the displaycolor is not changing. I am using your precompiled HomePoint v0.04rc1 binary. 

Question:
- do i use the key wrong?

{
  **"displayColorInverted": "true",**
  "wifi": "foo",
  "password": "foo",
  "login": "admin",
  "webpass": "foo",
  "mqttbroker": "foo",
  "mqttusername": "foo",
  "mqttpasswd": "foo",
  "timezone": "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00",
  "scenes": [{
    "name": "Wohnzimmer",
    "type": "Light",
    "icon": "livingroom",
    "devices": [{
      "name": "Side Light",
      "setTopic": "lights/hue/00:17:88:01:02:3c:2a:6d-0b/set/on",
      "getTopic": "hue/status/lights/Hue color lamp 1",
      "onValue": "true",
      "offValue": "false"
    }]
  }]
}

with kind regards and thanks for your support

michael

sieren, 2020-10-23:
Hey Michael,

indeed, youre using a string instead of a boolean:

```
"displayColorInverted": true,
```

Can you try using _true_ without the quotes and report back?

mherke, 2020-10-23:
Hi sieren,

i changed:
 "displayColorInverted": "true",
into:
"displayColorInverted": true,
without success. Also changed it to 'false' to see what happens, but without any changes of the display color.

...any ideas?

thanks for your support

michael

sieren, 2020-10-23:
Which M5Stack are you using and can you take a picture of the screen?
This code is only needed for newer IL displays used in the Core2 M5 units.

sieren, 2020-10-23:
Here's the cross reference https://github.com/sieren/Homepoint/issues/112

sieren, 2020-10-23:
Okay! I see what is happening. This fix hasn't actually been released in the binary yet. Can you try the version in the mentioned issue above and verify if that fixes it? I'll make sure to do a new release asap

mherke, 2020-10-24:
> 
> 
> Which M5Stack are you using and can you take a picture of the screen?
> This config is only needed for newer IL displays used in the Core2 M5 units.

Hi,
i am using the "ESP32 Basic Core IoT Development Kit":

[image]

mherke, 2020-10-24:
> 
> 
> Okay! I see what is happening. This fix hasn't actually been released in the binary yet. Can you try the version in the mentioned issue above and verify if that fixes it? I'll make sure to do a new release asap

Hi, i am running the firmware from the cross reference #112. Now the color inversion works. 

many thanks for your support

michael

closed 2020-10-24 by mherke