sieren/Homepoint #117: could not get mqtt value

ExPalm, 2020-10-27:
Hello Sieren,

thank you for this very nice project. I could run switches by using the ArduiTouch device but unfortunately I could not find a way to show values from my ESP sensors. I use Mosquitto as MQTT Broker. Devices are sending from FHEM and I can get the values with an ESP/OLED device but I could not find a way to confinger the config.json tho get the values on the AduiTouch. 
For example I can get values by using other devices by using the gettopic /draussen/temperature. When I put this topic in the
config.json like this  
       "type": "singleValue",
      "firstIcon": "temperature_small",
      "jsondata": true,
      "firstKey": "Temperature",
      "getTopic": "/draussen/temperature"
I get only 0 displayed. 
Do you have a hint for me?

Thanks

ExPalm

sieren, 2020-10-27:
In this particular case, what does `/draussen/temperature` actually return? Is it a JSON String (e.g. `{ "Temperature": 12 }`) or the raw value?
If it's the raw value, you need to remove `jsonData` and `firstKey` from the config.

koenvervloesem, 2020-10-27:
Are you sure your MQTT topic is `/draussen/temperature`? MQTT topics don't start with `/` like file paths.

ExPalm, 2020-10-27:
Here is a screenshot from MQTT Explorer. It looks for me like a raw value but when I remove  jsonData and firstkey only the icon is shown on the display. 
[image]

sieren, 2020-10-27:
It looks to me its a combination of what @koenvervloesem and I said.
The path of this topic is a bit odd to me too. Is there a blank space before the first `/` ?

ExPalm, 2020-10-27:
No there is no space in front. Here is a screenshot from FHEM where I defined the MQTT Bridge
[image]
Even when I take off the first / there nothing displayed only the icon

ExPalm, 2020-10-27:
Thank you I got it. It was the issue with the raw data. Thanks a lot and enjoy the coffee

closed 2020-10-27 by ExPalm

sieren, 2020-10-27:
Nice! Glad you got it sorted. What issue was it exactly? Always keen to learn in case someone else has this issue

ExPalm, 2020-10-27:
The hint to remove jsonDate and firstKey when only raw data are transfered was the solution