sieren/Homepoint #119: compile error MQTT_ANY

mwsun58, 2020-11-13:
during compiling i get this error:
homepoint/main/mqtt/MQTTConnection.cpp:64:14: error: 'MQTT_EVENT_ANY' was not declared in this scope
         case MQTT_EVENT_ANY:

when i am commenting out this part in MQTTConnection.cpp, compilation is ok and homepoint runs ok on the AZTouch.

My ESP-IDF Version is:
esp/esp-idf$ git describe --tags --dirty
v4.0-dirty

sieren, 2020-11-13:
Are you using the release version of ESP-IDF 4.0? (I suppose not, since it's marked dirty).
I had to add this recently because it was added to 4.0 seemingly last minute:
https://github.com/sieren/Homepoint/commit/dce1eeea02a79d6307d4a38a955b5397670939d9

closed 2020-11-13 by sieren

mwsun58, 2020-11-13:
Ja, richtig. War eine "dirty" Version. Hab nun mal 4.1 stable benutzt. MQTT_ANY ist nun gut. Aber nun gibt es was neues:

homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:102:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?

     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
                        ^~~~~~~~~~~~~
                        rtc_io_desc

scurfin, 2020-11-23:
> homepoint/components/arduino-esp32/cores/esp32/esp32-hal-gpio.c:102:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
> 
> ```
>  uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
>                     ^~~~~~~~~~~~~
>                     rtc_io_desc
> ```

Den Fehler hatte ich mit der 4.1 auch. Habe es dann mit dem letzten angezeigten Release von ESP-IDF v4.0.2 kompiliert und es lief ohne Fehlermeldung durch.

sieren, 2020-11-23:
Tracking this. I'll move the project on 4.1 in the coming days so we should hopefully no longer see those :)

sieren, 2020-12-04:
Looked into this. I will not support 4.1, the project needs v4.0.
This is because the Arduino Library used does not support 4.1, although it is currently being updated for ESP-IDF v4.2 which is in RC mode. I will update this project to 4.2 once there is a stable release.