sieren/Homepoint #120: no issue, only question about icons

mwsun58, 2020-11-13:
Hi,
is it possible to use own icons beside the icons in the data/ dir? 
What format they need to be? Or how can they be created? Is it only a JPEG Picture in the size 50x50?

Thanks

sieren, 2020-11-13:
yes you can. just follow the existing files in size and format, then use the name before the _active/_inactive endings in the configuration (see examples).

mwsun58, 2020-11-13:
Hi,
ich mach mal auf Deutsch weiter da ich gerade einen anderen Post gesehen habe wo es um die gleiche Frage ging. Ich habe mir zwei JPEG Files im Ordner data/ angelegt mit dem Namen desklamp_active und desklamp_inactive. Nach einem neukompilieren und flashen wird leider kein Icon angezeigt. Im Monitor kommt der Fehler: 
Couldn't resolve image /desklamp_active.jpg
I (15268) MQTT: MQTT_EVENT_DATA
Couldn't resolve image /desklamp_inactive.jpg
I (16493) MQTT: MQTT_EVENT_DATA

Die Lampe schaltet. 
Ich habe nochmal mit make clean alles auf Anfang gesetzt, neu kompiliert und im config.json nun als Icon 'livingroom' ausgewählt. Das Icon wird nach dem Neustart auch angezeigt, allerdings kommt im Monitor die gleiche Meldung, nur bzgl. livingroom: 
ouldn't resolve image /livingroom_inactive.jpg
I (12810) MQTT: MQTT_EVENT_DATA
Couldn't resolve image /livingroom_active.jpg
I (15268) MQTT: MQTT_EVENT_DATA

sieren, 2020-11-13:
Kann sein dass du `make spiffs_spiffs_bin` vergessen hast? Dadurch wird die SPIFFS Datenpartition neu erstellt wo sich auch die bilder drin befinden.

mwsun58, 2020-11-13:
Dachte ich auch erst, habe ich aber mitgemacht. :(

sieren, 2020-11-15:
Kannst du mal den abschnitt aus der config pasten?

closed 2020-12-04 by sieren