sieren/Homepoint #121: config.json speichern

papa-romeo, 2020-11-13:
Hallo,
ist inzwischen mehrfach aufgetreten dass sich die config.json nicht speichern lässt. Genauer gesagt, es werden nur Bruchstücke gespeichert, oder die config.json ist leer oder es erfolgt keine Änderung an der config.json nach dem Speichern und es erfogt die Ausgabe: "Configuration invalid!" mit den Zusätzen, dass die config oder die failsave nicht geparsed werden konnte, obwohl an der failsave keine Änderung vorgenommen wurde.
Gelegentlich kann man das System nochmal retten, wenn man die failsave.json in die config.json speichert und die Speicherung akzeptiert wird. 
Wenn nicht bleibt nur der Weg den ESP32 nochmals neu zu Flashen. 
Zuerst machte es den Anschein, dass dies nur dann vor kommt, wenn man in der config.json falschen Werte eingibt und speichert. Inzwischen hab ich aber  mehrfache Bestätigungen von anderen User, dass dies auch vorkommt, wenn eine lauffähige config.json, nochmal speichert wird.
Ich habe inzwischen mehrere ESP32 versucht, aber an diesen liegt es nicht.

LG

Papa Romeo

papa-romeo, 2020-11-15:
Hallo, 
soeben wieder pasisiert. Hatte eine lauffähige config.json und habe festgestellt, dass ich bei der Namensbezeichnung eines Touchfeldes ein Schreibfehler hatte. 
Statt "Werkstatt" hatte ich "Werstatt" eingegeben. Ich habe den Namen korrigiert und die config neu gespeichert.
Nach den Reboot kam dann die bekannte Fehlermeldung "Configuration invalid!" usw.
Nach dem Kopieren der failsave.json in die config.json, erschien nach dem Reboot das bekannte Gitterliniennetz,
jedoch jeder weiter Versuch, sei es die example.json oder die vorher funktionierende eigene config.json zu speichern schlug fehl. Es werden nur immer Bruchstücke gespeichert und am Schluß ist die config.json leer.
Ich habe fasst den Eindruck als würde der Speicher vollgeschrieben und könnte am Schluß nichts mehr aufnehmen.
Wenn man die "bin" dann neu flashed und die config.json einspielt funktioniert das System wieder.

LG

Papa Romeo

sieren, 2020-11-15:
Hey,

leider ein bekannter und sehr lästiger Fehler (wird hier getracked: https://github.com/sieren/Homepoint/issues/87 )
Bin leider bisher noch nicht auf die Spur gekommen wie man dies reproduzierbar macht, würde es aber auch gerne gefixed haben 

LG
Matthias

closed 2020-11-15 by sieren