sieren/Homepoint #122: Doesn't connect to the network

cph25, 2020-11-21:
My system
FritzBox 6490 Cable
WPA2 (CCMP)
2.4.and 5 GHz WLAN activ, both with the same ssid and passphrase

I entered the ssid and passphrase in the start-up webpage be copy/paste as well be typing, means the issue isn't caused by mistyping. Also the log shows the right ssid and passphrase.
I tried it with homepoint_espgeneric.bin as well with homepoint_espgeneric_full.bin, no difference

here a cut out of the log:

17:12:47.399 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:12:47.446 -> configsip: 0, SPIWP:0xee
17:12:47.446 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:12:47.446 -> mode:DIO, clock div:1
17:12:47.446 -> load:0x3fff0018,len:4
17:12:47.446 -> load:0x3fff001c,len:6736
17:12:47.446 -> ho 0 tail 12 room 4
17:12:47.446 -> load:0x40078000,len:12612
17:12:47.446 -> load:0x40080400,len:3788
17:12:47.446 -> entry 0x40080684
17:12:47.446 -> [0;32mI (65) boot: Chip Revision: 1[0m
17:12:47.446 -> [0;32mI (68) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
17:12:47.446 -> [0;32mI (40) boot: ESP-IDF v4.0.1-378-g26d7fb774-dirty 2nd stage bootloader[0m
17:12:47.446 -> [0;32mI (41) boot: compile time 17:43:20[0m
17:12:47.499 -> [0;32mI (42) boot: Enabling RNG early entropy source...[0m
17:12:47.499 -> [0;32mI (47) boot: SPI Speed      : 80MHz[0m
17:12:47.499 -> [0;32mI (51) boot: SPI Mode       : DIO[0m
17:12:47.499 -> [0;32mI (55) boot: SPI Flash Size : 4MB[0m
17:12:47.499 -> [0;32mI (59) boot: Partition Table:[0m
17:12:47.499 -> [0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
17:12:47.499 -> [0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
17:12:47.499 -> [0;32mI (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
17:12:47.499 -> [0;32mI (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
17:12:47.499 -> [0;32mI (92) boot:  3 ota_0            OTA app          00 10 00010000 00150000[0m
17:12:47.546 -> [0;32mI (100) boot:  4 ota_1            OTA app          00 11 00160000 00150000[0m
17:12:47.546 -> [0;32mI (107) boot:  5 spiffs           Unknown data     01 82 002b0000 0007d000[0m
17:12:47.546 -> [0;32mI (115) boot: End of partition table[0m
17:12:47.546 -> [0;32mI (119) boot_comm: chip revision: 1, min. application chip revision: 0[0m
17:12:47.546 -> [0;32mI (126) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x48744 (296772) map[0m
17:12:47.646 -> [0;32mI (232) esp_image: segment 1: paddr=0x0005876c vaddr=0x3ffb0000 size=0x03968 ( 14696) load[0m
17:12:47.646 -> [0;32mI (237) esp_image: segment 2: paddr=0x0005c0dc vaddr=0x40080000 size=0x00400 (  1024) load[0m
17:12:47.646 -> [0;32mI (239) esp_image: segment 3: paddr=0x0005c4e4 vaddr=0x40080400 size=0x03b2c ( 15148) load[0m
17:12:47.700 -> [0;32mI (253) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xda6bc (894652) map[0m
17:12:47.987 -> [0;32mI (546) esp_image: segment 5: paddr=0x0013a6dc vaddr=0x40083f2c size=0x1071c ( 67356) load[0m
17:12:47.987 -> [0;32mI (584) boot: Loaded app from partition at offset 0x10000[0m
17:12:47.987 -> [0;32mI (584) boot: Disabling RNG early entropy source...[0m
17:12:48.034 -> [0;32mI (585) cpu_start: Pro cpu up.[0m
17:12:48.034 -> [0;32mI (588) cpu_start: Application information:[0m
17:12:48.034 -> [0;32mI (593) cpu_start: Project name:     homepoint[0m
17:12:48.034 -> [0;32mI (598) cpu_start: App version:      v0.02b-26-gfbdd68e-dirty[0m
17:12:48.034 -> [0;32mI (604) cpu_start: Compile time:     Aug 18 2020 17:43:30[0m
17:12:48.034 -> [0;32mI (611) cpu_start: ELF file SHA256:  c7503aae3093667e...[0m
17:12:48.034 -> [0;32mI (617) cpu_start: ESP-IDF:          v4.0.1-378-g26d7fb774-dirty[0m
17:12:48.034 -> [0;32mI (623) cpu_start: Starting app cpu, entry point is 0x400812e8[0m
17:12:48.081 -> [0;32mI (0) cpu_start: App cpu up.[0m
17:12:48.081 -> [0;32mI (634) heap_init: Initializing. RAM available for dynamic allocation:[0m
17:12:48.081 -> [0;32mI (640) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
17:12:48.081 -> [0;32mI (647) heap_init: At 3FFBCBC8 len 00023438 (141 KiB): DRAM[0m
17:12:48.081 -> [0;32mI (653) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
17:12:48.081 -> [0;32mI (659) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
17:12:48.081 -> [0;32mI (666) heap_init: At 40094648 len 0000B9B8 (46 KiB): IRAM[0m
17:12:48.081 -> [0;32mI (672) cpu_start: Pro cpu start user code[0m
17:12:48.127 -> [0;32mI (692) spi_flash: detected chip: generic[0m
17:12:48.127 -> [0;32mI (692) spi_flash: flash io: dio[0m
17:12:48.127 -> [0;32mI (692) cpu_start: Starting scheduler on PRO CPU.[0m
17:12:48.127 -> [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
17:12:48.127 -> [E][esp32-hal-cpu.c:125] removeApbChangeCallback(): not found func=400F3034 arg=3FFB13A8
17:12:48.502 -> [E][vfs_api.cpp:22] open(): File system is not mounted
17:12:48.502 -> [E][vfs_api.cpp:22] open(): File system is not mounted
17:12:48.549 -> [0;31mE (1567) example: Failed to find Font[0m
17:12:48.549 -> I (1587) wifi:wifi driver task: 3ffde624, prio:23, stack:6656, core=0
17:12:48.602 -> [0;32mI (1587) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
17:12:48.602 -> [0;32mI (1592) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
17:12:48.602 -> I (1613) wifi:wifi firmware version: 9156a0d
17:12:48.602 -> I (1613) wifi:config NVS flash: enabled
17:12:48.602 -> I (1613) wifi:config nano formating: disabled
17:12:48.602 -> I (1615) wifi:Init data frame dynamic rx buffer num: 32
17:12:48.602 -> I (1620) wifi:Init management frame dynamic rx buffer num: 32
17:12:48.602 -> I (1625) wifi:Init management short buffer num: 32
17:12:48.602 -> I (1630) wifi:Init dynamic tx buffer num: 32
17:12:48.649 -> I (1634) wifi:Init static rx buffer size: 1600
17:12:48.649 -> I (1638) wifi:Init static rx buffer num: 10
17:12:48.649 -> I (1642) wifi:Init dynamic rx buffer num: 32
17:12:48.702 -> [0;32mI (1729) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0[0m
17:12:48.702 -> I (1730) wifi:mode : sta (24:62:ab:f3:a1:b0)
17:12:48.702 -> [0;31mE (1731) event: system event loop not initialized via esp_event_loop_init[0m
17:12:48.749 -> [0;32mI (1737) wifi station: wifi_init_sta finished.[0m
17:12:48.749 -> [0;32mI (1742) wifi station: connect to ap SSID:<CPH> removed<CPH> password:<CPH> removed<CPH>[0m
17:12:48.850 -> I (1858) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:12:48.850 -> I (1859) wifi:state: init -> auth (b0)
17:12:48.850 -> I (1865) wifi:state: auth -> assoc (0)
17:12:48.850 -> I (1871) wifi:state: assoc -> run (10)
17:12:52.878 -> I (5914) wifi:state: run -> init (2a0)
17:12:52.878 -> I (5915) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:12:52.925 -> I (5915) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:12:52.925 -> I (5919) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:12:52.925 -> I (5925) wifi:state: init -> auth (b0)
17:12:52.925 -> I (5932) wifi:state: auth -> assoc (0)
17:12:52.925 -> I (5938) wifi:state: assoc -> run (10)
17:12:56.995 -> I (9983) wifi:state: run -> init (2a0)
17:12:56.995 -> I (9984) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:12:56.995 -> I (9984) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:12:56.995 -> [0;31mE (9988) event: system event loop not initialized via esp_event_loop_init[0m
17:12:56.995 -> [0;32mI (9996) wifi station: retry to connect to the AP[0m
17:12:59.048 -> [0;31mE (12043) event: system event loop not initialized via esp_event_loop_init[0m
17:12:59.048 -> [0;32mI (12045) wifi station: retry to connect to the AP[0m
17:12:59.148 -> I (12165) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:12:59.148 -> I (12165) wifi:state: init -> auth (b0)
17:12:59.148 -> I (12169) wifi:state: auth -> assoc (0)
17:12:59.148 -> I (12175) wifi:state: assoc -> run (10)
17:13:03.180 -> I (16220) wifi:state: run -> init (2a0)
17:13:03.221 -> I (16220) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:13:03.221 -> I (16220) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:13:03.221 -> I (16225) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
17:13:03.221 -> I (16231) wifi:state: init -> auth (b0)
17:13:03.221 -> I (16237) wifi:state: auth -> assoc (0)
17:13:03.221 -> I (16243) wifi:state: assoc -> run (10)
17:13:07.264 -> I (20287) wifi:state: run -> init (2a0)

Thanks
Peter (CPH)

sieren, 2020-11-21:
Ive been testing this with a FritzBox myself and found it to be working. Can you double check all your WiFi Settings?

cph25, 2020-11-21:
I rebootet the box to avoid a hanging connection, no improvements.
I tried it again - same result.

sieren, 2020-11-21:
I'm afraid I don't know how to proceed from here, especially since the logs show the right SSID and Passphrase I suspect this may be an issue between the ESP32 SDK and your router configuration. I'm also not aware of any other reports regarding wifi so far. A quick search for "<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1 esp32" reveals that others are experiencing the same issue (unrelated to Homepoint).

closed 2020-12-04 by sieren