sieren/Homepoint #125: no issue, only a question (possibly a featurerequest)

sohnvonoff, 2020-11-26:
Hi,
there is the possibility to control the beeper with an MQTT subscribe?
That would be useful for an acoustic signal.

Thanks
Bernd

sieren, 2020-12-01:
Can you describe your actual use case a bit so I get a better idea of where this would be used?

sohnvonoff, 2020-12-01:
your project is really awesome. It hears and speaks MQTT. It can therefore be used universally for almost every home automation.
I have FHEM (https://fhem.de/) on a Raspberry Pi as home automation.
This means that 'Homepoint' together with 'AZ-Touch ESP' is an almost complete solution for a wall terminal.
I just miss the acoustic signaling for e.g. Alarms and warnings.
In your 'homepoint', every touch is confirmed with a short beep. Generating a beep is therefore possible!
For example, if you could use the topic 'Homepoint01/Alarm' value = 15, a beep with e.g. Generate 2kHz for 1.5 seconds and
with the topic 'Homepoint01/Warn' value = 25 a beep with e.g. Generate 1kHz for 2.5 seconds.

mike-gh0000, 2021-01-12:
thanks for your great project.
it works out of the box, mqtt broker + nodered on a raspberry and ESPEasy for the sensors.
the possibility to control the beeper with an MQTT subscribe would be great.
same idea as 'sohnvonoff'