sieren/Homepoint #127: Support for switching percentage-values (Dimming, Shutter, etc)

psi-4ward, 2020-12-18:
Hi,
are there any plans to support Devices which are controlled 0..100% like dimmers, shutters and so on?

Greets

sieren, 2020-12-19:
Not yet, but it crossed my mind. Would require a different UI though. I'll think about what would be a good approach

psi-4ward, 2020-12-19:
Not as trivial as a on/off switch. I think about a button which opens a new "fullscreen" page some arrows to step up/down and some buttons to: go-back, set a predefined value (OFF, 25%, 50%, 75%, 100%) or smth like that.

sieren, 2021-02-14:
Could need some more input to consider doing this.
What would the configuration look like for this? `/bla/device/brightness/' ? What's the common type for brightness? String? Integer? Float?
What is the common value range for brightness (`0..0.5...1`? `0...127...256`? ) and how would someone configure this.

htvekov, 2021-02-18:
Hi' Matt.

Both Z2M and HA uses integer 0-254 for brightness.
Using MQTT commands for Z2M lights or direct HA service calls with brightness: 0 not only dims light completely, but also turn off actual device as well.
Don't know how `brightness: 0` behaves in other environments.

Z2M `set` command: 

```json
topic: zigbee2mqtt/FRIENDLY_NAME/set
payload: {"brightness": VALUE}

```

Z2M `get` command:

topic: zigbee2mqtt/FRIENDLY_NAME/get
payload: {"brightness": ""}

Bulb will return current value in device state payload:

```json
topic: zigbee2mqtt/FRIENDLY_NAME
payload: {"brightness":254,"color_temp":370,"linkquality":107,"state":"ON","update":{"state":"idle"},"update_available":false}
```

Both HA and Z2m uses standard mired scale for setting bulbs color_temp.  The largest span for bulbs I've got, goes from 153 to 500 on the mired scale. Value is an integer as well.

Z2M command structure is completely identical. Only json key is `color_temp` instead of `brightness`

Ciao !