sieren/Homepoint #128: Numbers in JSON result in reboot

psi-4ward, 2020-12-18:
```
␛[0;32mI (70039) MQTT: MQTT_EVENT_DATA␛[0m
assertion "IsString()" failed: file "/Users/sierenmusic/Development/HomeControl/main/libraries/rapidjson/include/rapidjson/document.h", line 1773, function: const Ch* rapidjson::GenericValue<Encoding, Allocator>::GetString() const [with Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>; rapidjson::GenericValue<Encoding, Allocator>::Ch = char]
abort() was called at PC 0x40189b03 on core 0

ELF file SHA256: 18faed4d03d75adc

Backtrace: 0x4008685c:0x3ffd2460 0x40086c35:0x3ffd2480 0x40189b03:0x3ffd24a0 0x400ddaa0:0x3ffd24d0 0x400e008f:0x3ffd24f0 0x400e016d:0x3ffd2690 0x400e0279:0x3ffd2720 0x400e02f7:0x3ffd2790 0x400dda7d:0x3ffd27c0 0x400ddc51:0x3ffd27f0 0x400ddc89:0x3ffd2810 0x4011a1ea:0x3ffd2830 0x4011aac8:0x3ffd2850

Rebooting...
```

config:
```json
    "name": "Temperatur b",
    "type": "Sensor",
    "icon": "door",
    "devices": [{
      "name": "Temp b",
      "type": "singleValue",
      "jsondata": true,
      "firstIcon": "temperature_small",
      "firstKey": "val",
      "getTopic": "hm/status/Wandthermostat:2/ACTUAL_TEMPERATURE"
    },
```

mqtt-payload:
```json
{
  "val":3.1,
   "ts":1608319032692,
  "lc":1608306604934
}
```

sieren, 2021-01-05:
Thanks, that should not happen / it should accept all types in JSON.
I'll fix this soon.

sieren, 2021-01-05:
Which version were you running? And if so, can you give these binaries a try?

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5771704/homepoint_release.zip)

psi-4ward, 2021-01-05:
Sorry I do not have the hardware anymore to run the tests

htvekov, 2021-01-05:
Hi' @sieren

I have (out of curiosity) tested with a text string as JSON attribute (ON or OFF) to be read.
This also crashes and force Homepoint to reboot.
Probably identical issue.

I run the 'September special' m5stack_full.bin version.

Ciao !

htvekov, 2021-01-07:
Hi' @sieren 

Just downloaded and retested with Hp v0.05 (ESP_generic) and no crashes now reading non numerical JSON attributes 🙂🎉👍

Can read the state for eg. a light entity or whatever and it turns up nicely on the display.

One issue though:
In one of my tests I wanted to read the `state` of a light entity.
`state` attribute is returned, but its not the correct `state` attribute that is shown on the display. 
Displayed `state` attribute is found one level down in JSON string.
The returned attribute is the `state` for `update` key which is `idle`.
The expected `state` attribute shown is `ON`

```JSON
{"brightness":251,
"color":{"x":0.491,"y":0.39},
"linkquality":110,
"state":"ON",
"update":{"state":"idle"},
"update_available":false}
```

```JSON
{
    "name": "Test json data",
    "type": "Sensor",
    "icon": "door",
    "devices": [{
      "name": "test",
      "type": "singleValue",
      "firstIcon":"wattage_small",
      "jsondata": true,
      "firstKey": "state",
      "getTopic": "zigbee2mqtt/Office desk lamp"
    }]
  }
```

Also tested with a boolean attribute returned.
Homepoint didn't crash but just shows 0.0 for both `true` and `false` attributes.

```JSON
{"battery":100,
"illuminance":24,
"illuminance_lux":24,
"linkquality":55,
"occupancy":true,
"voltage":3005}
```

```JSON
{
    "name": "Test json data boolean attribute",
    "type": "Sensor",
    "icon": "door",
    "devices": [{
      "name": "test",
      "type": "singleValue",
      "firstIcon":"wattage_small",
      "jsondata": true,
      "firstKey": "occupancy",
      "getTopic": "zigbee2mqtt/Office"
    }]
  }
```

Ciao !

sieren, 2021-01-12:
I'll look into it :)

sieren, 2021-01-12:
I think you've found a bug. Can you verify by updating your firmware with the OTA part (or flash from the ground up) ?
Attached a build to test.

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5802739/homepoint_release.zip)

htvekov, 2021-01-12:
Hi' @sieren 

Just made a quick test on the M5stack using OTA file.(last tests was made with the ESP_generic files)
Didn't know what to check for - so I tested for both issues reported :-)

Still crashes reading a true or false boolean value from `occupancy` json key.
If i send a non valid boolean [truuuuue or falllse] or whatever it does crash. So json key is parsed by HomePoint for sure.
Retested on 'standard' v0.05 and found that it also crashed on M5_stack_OTA ? 
It didn't crash yesterday testing with ESP32_generic v0.0.5.  Very strange... 

`state` attribute read is also unchanged. Still shows `idle` attribute from `update: {"state":}` json key and not the expected `ON` attribute.

`MQTT publish: topic 'zigbee2mqtt/Office desk lamp', payload '{"brightness":227,"color_temp":370,"linkquality":92,"state":"ON","update":{"state":"idle"},"update_available":false}'`

Should I retest using the ESP32_generic files or are you confident that outcome should be identical ?

Ciao !

sieren, 2021-01-12:
Oh boy, yeah, apparently the M5Stack build failed after the build script refactoring in https://github.com/sieren/Homepoint/pull/132

Here's another build, this time specifically test for these two issues:
a) Should display "true" or "false" instead of crashing when a bool value is retrieved
b) Top level "state" key should be displayed

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5804851/homepoint_release.zip)

htvekov, 2021-01-12:
Hi' @sieren 

That was a quick fix ! 😁

Nothing much to say than both issues are now solved 👍

Issue a)
true or false displayed ✔

Issue b)
Top level JSON key is displayed (ON or OFF for that specific state) ✔

Regarding issue b)
It _could_ be nice  to be able to specify a nested JSON key attribute to be retrieved by adding the full path.
I have some sensors that provide nested JSON attributes with identical key names.
It's quite typical for combines temp/hum/pres sensors I believe.
I've no idea about what the syntax should be or what parsing capabilities you actually have in HomePoint ?
(Not really into nested JSON's 😉)
e.g. 

```json
{
    "name": "Test json data",
    "type": "Sensor",
    "icon": "door",
    "devices": [{
      "name": "test",
      "type": "singleValue",
      "firstIcon":"wattage_small",
      "jsondata": true,
      "firstKey":  "\"update\":{\"state\"}",
      "getTopic": "zigbee2mqtt/Office desk lamp"
    }]
  }
```

Thank you, Matt.
Much appreciated 🙋‍♂️

Ciao !

sieren, 2021-01-13:
Can you provide me an example of the JSON Data there?

closed 2021-01-13 by sieren

htvekov, 2021-01-13:
Hi' @sieren

Sure thing 👍
I have several sensors that use nested JSON's. I'm afraid that it's quite common.

Outdoor lux sensor:
```json
{"Time":"2021-01-13T14:39:33","BH1750":{"Illuminance":421}}
```

Greenhouse temp/hum/press. sensor:

```json
{"Time":"2021-01-13T14:37:30","SI7021":{"Temperature":3.4,"Humidity":96.8,"DewPoint":2.9},"BMP280":{"Temperature":4.1,"Pressure":996.3},"PressureUnit":"hPa","TempUnit":"C"}}
```

I have no issue with these at the moment.
I can always send the needed data from Home Assistant via MQTT if needed.

I agree that the Z2M /HA stuff is much more interesting for other users than me 😉👍 

CIao !