sieren/Homepoint #129: Building problem

xtimmy86x, 2021-01-01:
I have setup a machine with ubuntu, i have installed all the packet listed for building your project
but when i do the last command "make -j 8 flash" i result with an error

make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:193: esp-idf/main/CMakeFiles/__idf_main.dir/mqtt/MQTTConnection.cpp.obj] Errore 1
make[3]: *** Attesa per i processi non terminati....
In file included from /root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup_Select.h:22,
                 from /root/esp/homepoint/main/libraries/TFT_eSPI/TFT_eSPI.h:35,
                 from /root/esp/homepoint/main/touch/ButtonDriver.h:3,
                 from /root/esp/homepoint/main/config/Config.h:3,
                 from /root/esp/homepoint/main/model/Model.hpp:3,
                 from /root/esp/homepoint/main/AppContext.h:3,
                 from /root/esp/homepoint/main/wifi/CaptiveServer.cpp:1:
/root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup.h:156: warning: "TOUCH_CS" redefined
 #define TOUCH_CS 14
 
<command-line>: note: this is the location of the previous definition
In file included from /root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup_Select.h:22,
                 from /root/esp/homepoint/main/libraries/TFT_eSPI/TFT_eSPI.h:35,
                 from /root/esp/homepoint/main/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup.h:156: warning: "TOUCH_CS" redefined
 #define TOUCH_CS 14
 
<command-line>: note: this is the location of the previous definition
make[3]: uscita dalla directory «/root/esp/homepoint/build»
make[2]: *** [CMakeFiles/Makefile2:4159: esp-idf/main/CMakeFiles/__idf_main.dir/all] Errore 2
make[2]: uscita dalla directory «/root/esp/homepoint/build»
make[1]: *** [CMakeFiles/Makefile2:1676: CMakeFiles/flash.dir/rule] Errore 2
make[1]: uscita dalla directory «/root/esp/homepoint/build»
make: *** [Makefile:190: flash] Errore 2

sieren, 2021-01-05:
Hey Timmy,
please provide more info like the CMake output you've used to configure the project.
Also which ESP-IDF Version are you using?

xtimmy86x, 2021-01-08:
Hi, thanks for the reply,
the esp-idf is the version listed in the instructions ( 4.0 )
where do i find the cmake output?

thanks

xtimmy86x, 2021-01-08:
> Hey Timmy,
> please provide more info like the CMake output you've used to configure the project.
> Also which ESP-IDF Version are you using?

Hi, thanks for the reply,
the esp-idf is the version listed in the instructions ( 4.0 )
where do i find the cmake output?

thanks

sieren, 2021-01-09:
I dont see the actual error in there, just a bunch of wrnings about a redefinition that is normal though.
can you paste all of the output?

xtimmy86x, 2021-01-10:
root@ale-VirtualBox:~/esp/homepoint/build# make -j 8 flash
make[1]: Entering directory '/root/esp/homepoint/build'
make[2]: Entering directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
Scanning dependencies of target _project_elf_src
Scanning dependencies of target blank_ota_data
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Leaving directory '/root/esp/homepoint/build'
Scanning dependencies of target esp32_linker_script
make[3]: Entering directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
Scanning dependencies of target partition_table
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
[  1%] Generating project_elf_src.c
[  1%] Generating linker script...
[  1%] Generating ../../ota_data_initial.bin
[  2%] Generating ../../partition_table/partition-table.bin
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Leaving directory '/root/esp/homepoint/build'
[  2%] Built target esp32_linker_script
make[3]: Entering directory '/root/esp/homepoint/build'
Scanning dependencies of target __idf_app_trace
[  2%] Built target _project_elf_src
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Leaving directory '/root/esp/homepoint/build'
[  2%] Built target partition_table
make[3]: Entering directory '/root/esp/homepoint/build'
[  2%] Built target blank_ota_data
Scanning dependencies of target bootloader
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
[  2%] Creating directories for 'bootloader'
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
[  2%] Building C object esp-idf/app_trace/CMakeFiles/__idf_app_trace.dir/app_trace_util.c.obj
[  2%] Building C object esp-idf/app_trace/CMakeFiles/__idf_app_trace.dir/app_trace.c.obj
[  2%] Building C object esp-idf/app_trace/CMakeFiles/__idf_app_trace.dir/host_file_io.c.obj
[  2%] Building C object esp-idf/app_trace/CMakeFiles/__idf_app_trace.dir/gcov/gcov_rtio.c.obj
[  2%] No download step for 'bootloader'
[  2%] No patch step for 'bootloader'
[  2%] No update step for 'bootloader'
[  2%] Linking CXX static library libapp_trace.a
[  2%] Performing configure step for 'bootloader'
make[3]: Leaving directory '/root/esp/homepoint/build'
-- Found Git: /usr/bin/git (found version "2.27.0") 
[  2%] Built target __idf_app_trace
make[3]: Entering directory '/root/esp/homepoint/build'
Scanning dependencies of target __idf_cxx
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
[  2%] Building CXX object esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_exception_stubs.cpp.obj
[  3%] Building CXX object esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_guards.cpp.obj
[  3%] Linking CXX static library libcxx.a
make[3]: Leaving directory '/root/esp/homepoint/build'
[  3%] Built target __idf_cxx
make[3]: Entering directory '/root/esp/homepoint/build'
Scanning dependencies of target __idf_newlib
make[3]: Leaving directory '/root/esp/homepoint/build'
make[3]: Entering directory '/root/esp/homepoint/build'
[  3%] Building C object esp-idf/newlib/CMakeFiles/__idf_newlib.dir/heap.c.obj
[  3%] Building C object esp-idf/newlib/CMakeFiles/__idf_newlib.dir/locks.c.obj
[  3%] Building C object esp-idf/newlib/CMakeFiles/__idf_newlib.dir/poll.c.obj
[  4%] Building C object esp-idf/newlib/CMakeFiles/__idf_newlib.dir/pread.c.obj
[  4%] Building C object esp-idf/newlib/CMakeFiles/__idf_newlib.dir/pwrite.c.obj
[... 1618 lines cut ...]
                 from /root/esp/homepoint/main/touch/ButtonDriver.h:3,
                 from /root/esp/homepoint/main/config/Config.h:3,
                 from /root/esp/homepoint/main/model/Model.hpp:3,
                 from /root/esp/homepoint/main/AppContext.h:3,
                 from /root/esp/homepoint/main/wifi/CaptiveServer.cpp:1:
/root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup.h:156: warning: "TOUCH_CS" redefined
 #define TOUCH_CS 14
 
<command-line>: note: this is the location of the previous definition
[100%] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/libraries/TFT_eSPI/TFT_eSPI.cpp.obj
In file included from /root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup_Select.h:22,
                 from /root/esp/homepoint/main/libraries/TFT_eSPI/TFT_eSPI.h:35,
                 from /root/esp/homepoint/main/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup.h:156: warning: "TOUCH_CS" redefined
 #define TOUCH_CS 14
 
<command-line>: note: this is the location of the previous definition
make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:193: esp-idf/main/CMakeFiles/__idf_main.dir/mqtt/MQTTConnection.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/root/esp/homepoint/build'
make[2]: *** [CMakeFiles/Makefile2:4159: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make[2]: Leaving directory '/root/esp/homepoint/build'
make[1]: *** [CMakeFiles/Makefile2:1676: CMakeFiles/flash.dir/rule] Error 2
make[1]: Leaving directory '/root/esp/homepoint/build'
make: *** [Makefile:190: flash] Error 2

sieren, 2021-01-10:
The error looks like https://github.com/sieren/Homepoint/issues/119 to me.

xtimmy86x, 2021-01-11:
Thanks a lot, i've used the 4.0.2 version of esp-idf and all is ok

in my opinion you should correct the wiki and insert the 4.0.2 version

the command
git clone -b v4.0 --recursive https://github.com/espressif/esp-idf.git
would be
git clone -b v4.0.2 --recursive https://github.com/espressif/esp-idf.git

sieren, 2021-01-11:
Thanks! I've updated https://github.com/sieren/Homepoint/wiki/Build-from-source to reflect this

closed 2021-01-11 by xtimmy86x