sieren/Homepoint #137: Enhancement: Different icons for group entities

htvekov, 2021-01-16:
Hi' @sieren

It would be nice if different icons could be specified for each child entity in a group 😁
If nothing is specified then each entity enherits parent icon, else specific items are used.

Example below where group parent icon is `office2`,
Two of the child entities has tablelamp as icon and last entity has ceilinglamp as icon.

```
"scenes": [{
    "name": "Office",
    "type": "Light",
    "icon": "office2",
    "devices": [{
      "name": "Ceiling",
      "icon": "ceilinglamp",
      "setTopic": "homepoint/light.kontor_loft/set",
      "getTopic": "homepoint/light.kontor_loft/get",
      "onValue": "on",
      "offValue": "off"
    },
    {
      "name": "Desk",
      "icon": "tablelamp",
      "setTopic": "zigbee2mqtt/Office desk lamp/set",
      "getTopic": "zigbee2mqtt/Office desk lamp/get",
      "onValue": "ON",
      "offValue": "OFF"
    },
    {
      "name": "Bed lamp",
      "icon": "tablelamp",
      "setTopic": "zigbee2mqtt/Office bed lamp/set",
      "getTopic": "zigbee2mqtt/Office bed lamp/get",
      "onValue": "ON",
      "offValue": "OFF"
    }]
  }
```

Ciao !

sieren, 2021-01-18:
Can do :)

sieren, 2021-01-18:
[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5832202/homepoint_release.zip)

Here's a build that should do this :) Let me know if it works for you

htvekov, 2021-01-18:
Hi' @sieren

I'll test and get back to you ASAP 🚀🚀

htvekov, 2021-01-18:
Working perfectly 👍👏👏😎

[image]

closed 2021-01-19 by sieren