sieren/Homepoint #138: Decimals and Screen ON/OFF?

zejulio, 2021-01-18:
Hello,

Is it possible to choose decimals? I have power in watts, I can have 2345W, that makes no sense to display 2345.0, it's only using space in the screen...
I'm using M5 Stack.

I would also like to send a MQTT topic to turn screen ON or OFF, is it possible?

Thanks!

sieren, 2021-01-18:
For the decimals, I will look into this.

For the latter topic, I'm not sure how useful this is? Can you walk me through your particular use-case?
The screen already turns itself off with the screen-saver, why would you want to trigger this remotely?

zejulio, 2021-01-18:
Thanks.
I hope decimals it's easy, a true/false on config.json or a number on how many (I think 0 or 1 is enough...)

Regarding the screen, I'm starting to use this with Home Assistant, I will send MQTT topics directly to this device. 
I want the screen to be off if none is on the area (I have this information on HA), and I will display the current house consumption, and I want the display to turn ON automatically if the consumption is above a certain level (I can also do this on HA).
This is 2 examples on the usage to turn ON/OFF the screen with a MQTT topic.

Thanks for your great work.

closed 2021-01-18 by sieren

htvekov, 2021-01-22:
Hi' @sieren 

As an alternative to a hardware sensor on the ESP32 itself, a display on/off MQTT switch could be very usefull for HA users.
Display could, as @zejulio describes, be turned on/off from HA via room presence, BLE etc. sensors
As far as I understand HomePoint won't turn on display if just a sensor value is updated.
Only entity state changes on/off will trigger display to turn on, right ?

Ciao !