sieren/Homepoint #139: How to flash OTA?

zejulio, 2021-01-18:
Hello,

How to flash the OTA firmware?

Thanks!

sieren, 2021-01-18:
There's a button in the web ui for homepoint:

[image]

zejulio, 2021-01-18:
I press that button and got "page isn't working". I only folowed the steps on main page, I think they are not up to date cause there is no information about OTA, and how to flash the OTA bin file... (I'm sorry I'm new to ESP32 and don't know how to do it...)

sieren, 2021-01-18:
Ah, I see. Please click on "Choose File" first and select the OTA file. then press update firmware.
But good you brought it up, I feel like this needs better explanation.

zejulio, 2021-01-18:
Duh, now I understand!
I thought it should open a new page and then I select the file...
I think this being in the same "line" as "reload/reboot" makes it messy an can get some confusion...
Thanks!

sieren, 2021-01-18:
[image]
 

Made some changes for the next update :) Thanks!

zejulio, 2021-01-18:
I would push update firmware below the configuration, since when you are configurating you do not want to scroll down too much (OTA takes too space on the page), or move it to a different page (like tasmota guys do).
But indeed this way is better and no confusion now.

sieren, 2021-01-18:
Good call! Will do that

zejulio, 2021-01-18:
So when will it be a new version? :)

sieren, 2021-01-18:
Soon :)

closed 2021-01-18 by sieren