sieren/Homepoint #145: Support for M5Stack Core2 planned?

kraus-wagner, 2021-01-30:
Will you support the new M5Stack Core2?

sieren, 2021-02-01:
Hey, the Core2 should already be supported. 
All it needs is to add `"displayColorInverted": true` to the configuration file since it's display is wired up a bit differently.

cerietke, 2021-02-01:
Interesting, not sure you're aware but the Dutch magazine Computer Totaal mentioned your work and they suggested the Core2 implementation wasn't yet ready (https://computertotaal.nl/artikelen/internet-thuis/zo-maak-je-een-eenvoudig-dashboard-voor-je-domoticasysteem/). This is why I got myself a Core. The Core2 with its touchscreen would be way cooler.

I actually found that to get the colors in their article I also had to set the displayColorInverted to true on the Core.

sieren, 2021-02-01:
Oh, wow. I wasn't aware of this! I guess it's time I order myself a Core2 and tidy this up nicer!

cerietke, 2021-02-01:
Ah, cool. I actually found their description of how to get things up and running pretty helpful at times, it felt slightly more detailed than your own (their demographic includes less experienced users as well). I'm a Product Owner, not a die-hard techy, though I like fiddling around with things, possibly it would be useful if I write some thing up regarding my experience as feedback?

sieren, 2021-02-01:
Absolutely! There's a Wiki here on Github which I sadly have neglected way too much where this could be very useful for!

cerietke, 2021-02-01:
Where do I send it?

koenvervloesem, 2021-02-01:
> Ah, cool. I actually found their description of how to get things up and running pretty helpful at times, it felt slightly more detailed than your own

Thanks, that was the intention of my article, to help non-techies with getting Homepoint up and running :-) Nice to hear that this worked.

In fact, I tried flashing Homepoint to the M5Stack Core2, but the device didn't boot (or at least I didn't see anything), and I didn't have the time then to investigate this further. Maybe the display code needs more changes than just inverting the colors, or maybe it's something else. In the mean time, I have been using my M5Stack Core 2 for another purpose, so I haven't been able to look at it further.

But I agree with the touch screen it would be a very cool device for Homepoint!

sieren, 2021-02-01:
@koenvervloesem Thanks for the article! :)
I will probably end up creating a special _bin file for M5Stack Core2 after all to make it easier for newcomers.
If I had unlimited resources I'd probably even get rid of the JSON Editor in favour of a modular Web UI.

@cerietke you can mail it to me at matthias (at) s-r-n.de

sieren, 2021-02-01:
Mijn nederlands is unfortunately not zo goed als dat ik dit en engels vertaalen kan (als en duitser ;)).
I think everyone would probably benefit from these step by step instructions. Super happy to host them and put them prominently on the page first.
I'll make sure to link to the article in the Wiki as well.

kraus-wagner, 2021-02-01:
@koenvervloesem Good article! I read it with google translator. ;-)
@sieren Great project!

I was looking for a touchscreen for an ESP32 or ESP8266 to send mqtt signals to my mosquitto and ioBroker.
It should fit in a wall switch. A common system size in Germany is 55mm (Gira...).
So i found the M5Stack Core2. A perfect gadget. ESP32, touchscreen, perfect size.
First steps in UIflow and Arduino IDE followed. Then i found your project. It is so amazing.
Everything can be controlled by a fingertip with mqtt. I am sure i will love it when it works on the Core2.

cerietke, 2021-02-01:
Cool that you're also checking in here Koen, I was not aware of homepoint nor the M5Stacks and it covers a lot of nice usecases I was looking for solutions for.

Better stick with English or German ;-). I'll mail you with feedback.

As an experiment I essentially turned the M5Stack Core into a lights and music remote control for my bedroom (receiver is in another room), but I was also thinking about the wall switch replacement option.

sieren, 2021-02-01:
Thanks! 

For Wallswitches, I vaguely recall there was a Kickstarter Project trying to achieve this too:
https://www.kickstarter.com/projects/159751026/23-smarthome-touch-display-w-wlan-for-55mm-switch-series

kraus-wagner, 2021-02-01:
Yes i found that project too. But the Core2 is ready to go. I ordered 2 from digikey.de for 39 Euro each.
https://www.digikey.de/product-detail/de/m5stack-technology-co-ltd/K010/2221-K010-ND/13151126
If your order is over 50 Euro shipping is free. They ship from USA with UPS in 3-4 days. Customs is in the price included.
So i paid 78 Euro total for 2 Core2. Shipping to my door.

sieren, 2021-02-03:
Received one today, but this will take longer than expected. Can't even get the screen to work right now.
And I need to write a new touch driver from scratch for this

kraus-wagner, 2021-02-03:
I use the Arduino IDE. There i have to change the library from "#include <M5Stack.h>" to "#include <M5Core2.h>" for example to run most code for the Stack on the Core2. I dont know how it works with the Espressif IDE.

sieren, 2021-02-03:
Yeah. But HomePoint supports multiple platforms with multiple capabilities and introducing dependencies on specific platforms is to be avoided. For it to work on all of them the code has to be generic and abstracted enough, and there come the challenges when a device introduces new components (like Core2 using a different touch-screen handler and moving certain TFT related capabilities to a separate power management chip)

kraus-wagner, 2021-02-04:
Ok i see. It is much more work than changing a library. But i hope you like your Core2 and will have time to make homepoint run on it.
✅

cerietke, 2021-02-05:
I got one as well, so hoping to see homepoint work on it at some point in the future.
Paid 34+2.50 shipping at tinytronics.nl.

sieren, 2021-02-05:
[image]

Got the display up and running today, next up Touch Driver support. 

If you feel like helping out but lack the programming skills. the best way is a small donation (there's a link at the end of the front page README) as I bought this device just to add compatibility for all the requests.
Personally I don't really have that much need for it.

jpiscaer, 2021-02-06:
Yay! Sitting here at virtual FOSDEM and tinkering with HomePoint and the Core2. Display won't turn on, version 0.6 with displayColorInverted set. Going to wait for a pre-compiled binary core the Core2.

sieren, 2021-02-06:
[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5937726/homepoint_release.zip)

Here's a first beta. Sadly the images dont look super pretty. The Core2 library is sadly also a far-cry from the Core1 one (https://github.com/m5stack/M5Core2/issues/33). Ultimately I might have to tear the Core2 one apart, which might mean lots of more work until things look as good as with other ESP32 devices.

kraus-wagner, 2021-02-06:
I uploaded the new core2 binary and have access to the web interface. I change the config file, so the Core2 is in my wifi network.
But the screen stays dark. I added "displayColorInverted": "true", to the config file and rebooted. Screen stays dark.
Any tips?

sieren, 2021-02-07:
Sorry. there was a bug in the build script that leads to the binary actually being the same as M5Stack (non-Core2)
[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5939033/homepoint_release.zip)

kraus-wagner, 2021-02-07:
Now it works! I made an OTA update and a reboot. 
"displayColorInverted": "true", must be removed from config if previously has been added. 
Now i can do a little testing ;-)

sieren, 2021-02-07:
Alright, just implemented proper JPG drawing, so no more artefacts. Did a new release too: https://github.com/sieren/Homepoint/releases/tag/v0.07

If you enjoy this work, please consider donating so I might be able to recoup the Core2 which I bought for this ;)

jpiscaer, 2021-02-08:
Trying to flash 0.07, but after flashing the full (and the OTA), I hear the device constantly booting with a typical 'reset' tick sound. It physically makes noise: ticking and sometimes a high-pitch whining. Screen won't turn on, and no 'homepoint' access point. If I flash back the factory test program with m5burner, the device works. Any idea what's happening? Factory test indicates no hardware problems.
`esptool.py write_flash -fs 4MB -fm dout 0x0 homepoint_m5stackcore2_ota_update.bin`

kraus-wagner, 2021-02-08:
@jpiscaer 

Hi, i first flashed the core2_full binary.
Then connected via the browser to 192.168.99.1 and entered my wifi settings. 
Next reconnection via the new ip in my network.
Last i made the OTA update also via the browser frontend.
(screen stays dark during all the time)
After a reboot the screen should work.

sieren, 2021-02-08:
Oh boy, this looks like a bug. I'll take a look later today.

sieren, 2021-02-08:
Phew, sorry for the annoyance: https://github.com/sieren/Homepoint/releases/tag/v0.07.1

jpiscaer, 2021-02-08:
Yes, this one boots. Thank you Matt! I appreciate your hard work

cerietke, 2021-02-14:
I had difficulty with connecting to it over WiFi. Both initially for the WiFi config as well as after being able to connect and set it up. I found sometimes it would help to restart. It seems flashing it a 3rd time it's stable as I have been able to load my config. Is there some way to confirm if there might be something up with my hardware?

P.s. donation made

sieren, 2021-02-14:
Mhmm, not sure. Maybe with the Factory tool it came with? 
I haven't had any issues with it so far. It does sound a bit dubious though, I haven't made any changes to the WIFI Part of it at all, its the same across all esp32 builds.

Thank you for the donation! appreciate it a lot 🙏☺️

jpiscaer, 2021-02-14:
i have seen the same behavior consistently: in the first couple of minutes after boot, WiFi is unstable, but it stabilizes afterwards. WiFi confirmed to be working with other firmware and factory test.

sieren, 2021-02-15:
I assume you mean it's crashing when you talk about "unstable" ? Thats a bit vague for me, but i did see some reboots when setting it up initially just now. I'll investigate.

jpiscaer, 2021-02-17:
Yes, with unstable I meant reboots, plus the dropping of connection in the first minutes after boot (seen by the wifi icon on the display)

dummylabs, 2021-03-18:
I can also confirm that m5core2 freezes after a couple of hours with the Homepoint running on it.

heartkingz, 2021-04-08:
Great project! I'm trying to use Homepoint but i am also experiencing WiFi issues. Most of the time it won't connect, it is very random. Sometimes a reload helps, sometimes a reboot and after that it connects very fast and i can use it without any issues. What i have noticed is that when you go to the web interface of the device it connects fairly fast after that.. 
Another thing i'm facing is that you have to press pretty long before a button responds, is that by design? I'd love to see a possibility to simply press short. Thank you again it's a great project!

sieren, 2021-04-23:
Just a quick update on this. I've picked up work on https://github.com/sieren/Homepoint/issues/155 ... it'll take a while but all platforms will benefit from this. I've started with the Core2 as the base development platform before expanding to the other ones. So far everything is stable, but rewriting the whole UI will take a bit more time.

Will keep you posted.

https://user-images.githubusercontent.com/5174440/115853174-2e293900-a429-11eb-84f7-91644a59c19c.MOV

sieren, 2021-05-04:
https://user-images.githubusercontent.com/5174440/117072109-4570fc00-ad30-11eb-8a80-f7f6234a55b0.mov

Expect a first beta in the coming days, the only thing currently missing is the sensor views and a few refinements before i can put a binary in this thread for some testing. Pretty happy with the new UI and being able to swipe between pages.

sieren, 2021-05-07:
Here's a very first beta of HomePoint 2.0.
Most of the original functionality should exist. Navigation between screen through swiping, long press to enter a scene with multiple devices, single tap to activate the whole scene.
Top right to exit a scene. Settings button currently does nothing.
Overall still rough around the edges.

Flash Example: `esptool.py --baud 921600 --port /dev/cu.usbserial-0202E890 write_flash -fs 4MB -fm dout 0x0 homepoint_m5stackcore2_full.bin`

[homepoint_m5stackcore2_full.bin.zip](https://github.com/sieren/Homepoint/files/6441172/homepoint_m5stackcore2_full.bin.zip)

heartkingz, 2021-05-08:
Just flashed the latest bin file tot the core2. Looks really good! I have to play a bit more to 

may I be so free to make suggestions:

- I'd love to see a battery indicator (percentage)/charging state on top 
- Love to see the two buttons on the side to have functions. The power button to power it on and off, the other one to put it     into deep sleep to preserve battery. 
- IP address to the settings icon top left corner 
- Maybe separator lines between the different rooms 

Again, thank you for your work!

sieren, 2021-05-08:
fThanks or your feedback. I'll definitely keep that in mind. It's still a long way of cleaning up and refining the current beta before I can think of adding new features. Let me know if you run into some weird or off behaviors.

cerietke, 2021-05-22:
Seems to work much better than the last version I tried a few months ago. The suggested command didn't work for me, but I got it loaded with "esptool.py write_flash -fs 4MB -fm dout 0x0 D:\Homepoint\homepoint_m5stackcore2_full.bin".

- Did something change with the icon functionality? Can't seem to get it to show my own icons the way it does on my core 1 with older installation. I used to just upload them through the web interface.
- Loving the scrolling between screens, much better, also better for a mix of core 1 and core 2 as the screens will both be able to contain 6 items.
- First connection for config was a bit slow, but after that it seems to be working fine.
- I may need to look into my config, but everything I have on it seems to be a group rather than a button. Definitely assuming that's on me.
- Would be nice if "back" took you back to the screen you were on.

sieren, 2021-05-22:
Yeah, the images now need to be on a fixed 16 bit color scale and be converted to binary files through https://lvgl.io/tools/imageconverter. The ending is now ".bin", loading and decoding JPGs/PNGs on the fly was just too slow, especially with the animations.

Entering a group is through a single tap, long-pressing triggers the whole group.
You can leave the screen by tapping on the top (little X), but I need to tweak this a little, its definitely too fiddly on the small Core2 screen.

cerietke, 2021-05-22:
I just noticed, if you have 6 devices at the lower level you only have the tiny top left 'x' to get back up, can be hard to hit. Was just going to post that, but I see you're aware. How about swipe up to go back?

sieren, 2021-05-22:
Maybe I'll keep the back button at the end. Might be worth it.
Detecting Swipe Gestures (ie. swipe up) isn't trivial with lvgl, but i'll take a look..

Yes, I guess you can long-press it for now to toggle it on/off, but I might add some logic to ensure a single tap is sufficient if the group only contains a single device.

cerietke, 2021-05-22:
I'd much prefer no back button as it gives an extra space. Obviously I don't know how difficult the swipe up could be. The icon could certainly be a bit bigger. Or how about clicking anywhere on the top bar to go up?

That would work for my scenario, or possibly a way to specify adding a device at the top level that isn't a group? Might not need extra logic then and may prove more flexible?

I'd really miss the screensaver, ideally I'd be able to put it at like 5 or 10 seconds. Also, using those settings I was unable to create an image file the core2 could work with, hence the question how I could get that fixed 16 bit color scale?

fvanroie, 2021-05-22:
You should be able to detect the [swipe gesture and direction](https://github.com/lvgl/lvgl/issues/1650#issuecomment-668594714) in your event handler like this:

```cpp
void event_handler(lv_obj_t* obj, lv_event_t event)
{
    if(event == LV_EVENT_GESTURE) {
        lv_gesture_dir_t dir = lv_indev_get_gesture_dir(lv_indev_get_act());
        switch(dir) {
            case LV_GESTURE_DIR_LEFT:
                break;
            case LV_GESTURE_DIR_RIGHT:
                break;
            case LV_GESTURE_DIR_BOTTOM:
                break;
            default:
                break; // TOP
        }
    } else {
      // other events
    }
}
```

sieren, 2021-05-22:
Awesome..I'll add this. But will keep the x at the top too. As nice as this gesture is, discovery might be an issue.

sieren, 2021-05-23:
@cerietke you have to make sure your images are of PNG & 16 bit indexed colors.
Homepoint 2.0 is using black (#00000) as a chroma-keyed background, meaning black will become transparent.

Here are a two images from the upcoming release:
[image]
[image]

cerietke, 2021-05-23:
Ha, I started from JPG. Still unsure how to get paint.net to output as 16 bit indexed, but I'll try later and report back.

cerietke, 2021-05-23:
Though the results look quite terrible I have been able to get it to display an image. 

[image]

This came out as dark grey and black.

sieren, 2021-05-23:
Yes, because it's saved as RGB and not Indexed Colors.

Try running this through the converter:
[image]

cerietke, 2021-05-23:
I'm sure it can be done. But what do you use to get to the correct image? I cannot find the correct option in paint.net. I am happy to use something else. Or am I using the wrong settings on https://lvgl.io/tools/imageconverter ?

sieren, 2021-05-23:
Both GIMP and Photoshop can get the job done. (e.g. https://docs.gimp.org/2.10/en/gimp-image-convert-indexed.html )

cerietke, 2021-05-23:
Indeed appears like paint.net doesn't have an indexed mode.

cerietke, 2021-05-23:
Trying gimp, when I copy your file it works. When I start from my png it does not.

I do image -> mode -> indexed -> maximum number of colors (16) -> convert, then I export to png (leaving the default settings). There must be something you're doing different than I am.

Besides the images, any insight on the screensaver? I'd plan to use this as a nightstand device, so screen always on would be a no-go. In fact it would be very nice to be able to disable the LED at the bottom and any other lights as well.

sieren, 2021-05-24:
Yes, I converted it to Greyscale before going indexed just to make sure there's no colors left in it.
Could be worth a shot?

As for the screensaver: It's coming back at some point. It was a lot of work transitioning to this library and I'm not done yet, I'll do the screensaver pretty soon after cause I'm facing the same problem as you.

sieren, 2021-05-24:
Oh, one thing I forgot. I used the "Fill" tool once on the black background of your image since it wasnt really entirely black before. there were a few greyish pixels.

cerietke, 2021-05-24:
Tried grayscale -> indexed, also tried with a save in between. Still doesn't show the picture.

I also see a black / white option in indexed, I am thinking to go for that to make absolutely sure it's only black and white.

[15 more replies]