sieren/Homepoint #146: M5Paper support

tigattack, 2021-02-03:
It would be great if support for the [M5Paper](https://m5stack.com/products/m5paper-esp32-development-kit-960x540-4-7-eink-display-235-ppi) could be added.

littlbee, 2021-02-04:
+++ on this. I have a M5Paper on my desk just only waiting for this project to support it !

jpiscaer, 2021-02-06:
Yes please!

sieren, 2021-02-06:
Happy to do this, but might have to rely on donations to afford buying one for this project.

littlbee, 2021-02-06:
Just did, 10€ donated.

jpiscaer, 2021-02-06:
@sieren donated a coffee just now. Thanks for all your work!

sieren, 2021-02-14:
Still quite a bit short of getting one financially.
Seems like theyre around 70-80 EUR/Unit in EU?

Most stores in Germany seem to be sold out. Any hints?

littlbee, 2021-02-14:
Got mine from digikey, 56€ without shipping, still in stock : https://www.digikey.be/products/en?keywords=M5paper

sieren, 2021-02-14:
It's 67 in DE with availability saying "Backorder" once its in the cart :/  https://www.digikey.de/products/en?keywords=M5paper

littlbee, 2021-02-14:
Not sure what to say... I paid 56 on digikey.be (not de). Indeed seems on backorder. Banggood still has few on stock but a bit more expansive. This one here (https://www.distrelec.biz/en/m5paper-540x960-ink-touch-display-core-m5stack-k049/p/30186369) puts it at 59€ and seems to be in stock

sieren, 2021-02-14:
Lucky you 😄

Well, I guess I'll have to wait for some more donations anyway.
I just got a Core2 for the sake of this project (on another Issue) and haven't even managed to recoup this one. It's open source after all, and looking at the low number of code contributions or other people adding new platforms, donations is currently the only way for me to come out of this at least at 0 financially.

megawubs, 2021-02-24:
@sieren I've also got one. I was thinking it might be fun to build something together? I want to learn programming esp stuff and micro controllers in general. I do have programming experience, as I'm a full time web developer. But no C, or anything related. Its a whole other realm. What do you think of this?

sieren, 2021-02-24:
@megawubs I like this idea! I'd definitely be up for that.

I'm currently working on https://github.com/sieren/Homepoint/issues/155 (rewriting the Homepoint front-end using LVGL).
Hopefully that will make supporting other displays and touch/button interfaces easier in the long run, but it will take some time until this is done.

megawubs, 2021-02-25:
How can I contact you @sieren?

martinschki, 2021-03-23:
Yes please!
m5paper support would be awesome

diewoutman, 2021-04-14:
@sieren I am in the same boat as megawubs. I've also got a M5Paper.
Would love to learn more about the esp's and micro controllers.
If i can help anywhere just let me know!

jamalex, 2021-11-18:
Donated €60 -- this would be amazing!