sieren/Homepoint #149: White screen on M5stack Fire

hassio-jaro, 2021-02-10:
First of all I would like to thank you Matt for this great piece of software.

Sadly the latest releases v0.07 and v0.07.1 are not working anymore on my M5stack Fire.
Going back to v0.06 everything is fine.

With latest version I first noticed, that display turns 180°.
Additional parameter in config.json "screenRotationAngle": 1 solved this, but after a few minutes of operation the display turns to white screen. Reboot doesn't solve the problem. 
Only flashing v0.06 recovers normal operation again.

sieren, 2021-02-10:
How annoying! I've indeed found a few bugs when compiling for M5Stack that got introduced while working on M5Stack Core2. Can you give this new build a shot? Flashing over OTA should suffice.

[homepoint_release.zip](https://github.com/sieren/Homepoint/files/5959064/homepoint_release.zip)

hassio-jaro, 2021-02-10:
> How annoying! I've indeed found a few bugs when compiling for M5Stack that got introduced while working on M5Stack Core2. Can you give this new build a shot? Flashing over OTA should suffice.
> 
> [homepoint_release.zip](https://github.com/sieren/Homepoint/files/5959064/homepoint_release.zip)

That solves the problem.
Just don't see any changes in the Web Interface as mentioned as feature, but that is a minor problem.
Also it would be nice to have the latest software version displayed somewhere in the web interface.

Keep on the good job...

sieren, 2021-02-10:
Thanks for checking! The changes in the web interface require a full reflash since the HTML is hosted on the data partition

closed 2021-02-10 by sieren