sieren/Homepoint #155: Rewrite Homepoint using LVGL

sieren, 2021-02-24:
I've started rewriting Homepoint using LVGL, which will help support more platforms - beyond ESP32 - in the long run and possibly enable Raspberry Pi support.
Spawned from https://github.com/sieren/Homepoint/issues/130

https://github.com/lvgl/lvgl

sieren, 2021-05-04:
https://user-images.githubusercontent.com/5174440/117072300-85d07a00-ad30-11eb-93a9-c5626e8fe60a.mov

Closing in on being able to release a first beta and started developing on master again.

sieren, 2021-05-07:
Beta versions for AZTouchMod and M5Stack Core 2

[homepoint_m5core2_aztouchmod_betas.zip](https://github.com/sieren/Homepoint/files/6443351/homepoint_m5core2_aztouchmod_betas.zip)

Girgitt, 2022-01-17:
Initial impressions on M5Core2 are very good. The lvgl-based version seems stable (at least no unexpected reboots experienced). 

However with the screen layout copied from the example config the initial screen is gray with IP address appearing after a couple of seconds. To get proper screen content one must tap to change screen and then move back manually triggering redraw/refresh.

Do you have plans to continue development? 

Your approach to navigation seems better compared to the much simpler code of cfGUI library which I extended to support M5core2 but any other development is probably worthless given how functional and eye-candy your design is.

sieren, 2022-01-18:
Thanks for your assessment @Girgitt :) There are still tons of bugs in there and the code is still somewhat messy as I've been trying to do performance improvements to increase FPS.

It's true, I haven't contributed much to the repo recently, but I have not abandoned the project. It's just that I had very little time to work on side projects in the past 6 months but I hope to return sometime soon.

Girgitt, 2022-01-18:
That's some good news :) Thank you for reaching back. I wish my Cpp skills were much better to be able to provide some valuable contribution but who knows, maybe it's a good enough reason to at least try when M5core2 becomes the supplemental HMI to my Homematic IP network in the next couple of months.

Girgitt, 2022-08-27:
Hi @sieren would find time to update/add building steps for homepoint-2.0-lvgl branch?
 
Instructions from wiki are obsolete as esp-idf version changed to v4.4 (at least this is what I found based on arduino-esp32 submodule's history for version committed to the homepoint-2.0-lvgl branch) but even with an updated esp-idf the compilation using /scripts/build_all.sh fails due to error: 
homepoint/components/ESPAsyncWebServer/src/WebAuthentication.cpp:74: undefined reference to `mbedtls_md5_starts'

Maybe some specific tag of esp-idf should be used?

(There are other issues like code from /scripts/merge_bin_esp.py being written in py2 while esp-idf uses py3 with virtualenv which makes the environment difficult to initialize. Fortunately 2to3 seems to convert merge_bin_esp.py just fine and /scripts/build_all.sh works with py3.8),

greiner11, 2023-07-28:
I concur with @Girgitt opinion. I am also enthusiastic about the project - it's a pity that it has not been developed further for some time.
It would also help me if I could compile it myself. However, I also failed with several errors.
The branch from Homepoint 2.0 has a VSC-Workspace. Is it possible to compile the code with Visual Code and the extension espressif (esp-idf extension)?

Girgitt, 2024-01-15:
So I tried to re-visit the project after a year and OMG what an adventure. First of all, despite what I wrote earlier, it is impossible that esp-idf version required to compile it is v4.4.1. IT IS NOT. it is something between 4.4.0 and 4.4.1. The proper version should be the esp-idf master at the time of arduino-esp32. Good luck in finding that - there is a version in submodule components/arduino-esp32 but it is Sieren's fork. Anyway by trying to use esp-idf v.4.4.1 and following error about missing reference to mbedtls_md5_starts i found commit that defines mbedtls_md5_starts_ret:
```
zbig@TX1320-1:~/repos/priv/esp/esp-idf_4_4_1$ git log -S "define mbedtls_md5_starts"
commit e87c64dc4443d973b784c9674a79259493abf9c0
Author: liuhan <liuhan@espressif.com>
Date:   Tue Apr 6 14:21:28 2021 +0800

    mbedtls: Add CONFIG_MBEDTLS_ROM_MD5 to use ROM MD5 in mbedTLS
zbig@TX1320-1:~/repos/priv/esp/esp-idf_4_4_1$ git show e87c64dc4443d973b784c9674a79259493abf9c0 | grep esp_md5_init
+int esp_md5_init_ret( mbedtls_md5_context *ctx );
+void esp_md5_init( mbedtls_md5_context *ctx );
+#define mbedtls_md5_init                        esp_md5_init
+#define mbedtls_md5_starts_ret                  esp_md5_init_ret
+int esp_md5_init_ret( mbedtls_md5_context *ctx )
+void esp_md5_init( mbedtls_md5_context *ctx )
+    esp_md5_init_ret(ctx);
```
**One commit earlier 1f54d17503ff17d830de75eed4eb47bc6a3aafc3 was a good candidate and it works** - at least Homepoint from branch homepoint-2.0-lvgl compiles. To avoid issues one needs to: 
esp-idf# rm -rf ~/.esspressif 
esp-idf# git checkout 1f54d17503ff17d830de75eed4eb47bc6a3aafc3
esp-idf# git submodule update --init --recursive
(make sure to install python3 requirements from install guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html )
esp-idf# ./install.sh
esp-idf# . ./source.sh

Then (after setting up python2 in your environement and modifying build_all.sh to use python2 instead of python and optionally commenting out steps for platforms you don't need) compile Homepoint with 
homepoint/scripts/# ./build_all.sh

At this point I am not sure if the resulting build works but size seems correct. Time to look for USB A to C cable.. 

Before that I tried many esp-idf versions and here is other way that resulted in compilation but M5Core2 did not start after flashing with esptool: 
1. checkout esp-idf to v.4.4-beta1
2. change Homepoint/components/ESPAsyncWebServer/src/WebAuthentication.cpp:74 mbedtls_md5_starts to mbedtls_md5_starts_ret
3. add values from https://github.com/espressif/arduino-esp32/issues/5925#issuecomment-981112887  to esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld