sieren/Homepoint #158: Version of homepoint in configuration webpage?

ghosty-be, 2021-02-25:
Was thinking if it would be possible to write the version of the homepoint firmware into a file on the spiffs (for example version.txt) so when you open the webpage that it can use that file to list the version on the configuration webpage somewhere?

sieren, 2021-02-25:
SPIFFS will not be updated during/after a firmware update, therefore thats not possible.
Showing the version is a good idea though, will do some point in the future.

ghosty-be, 2021-02-25:
but from your running code on bootup it should be possible to write a file in the spiffs, no?
And I am not familiar with the ESP-IDF development environment you used to develop this... I mostly just work with arduino IDE...
Or maybe I shouldn't be so lazy and setup a full dev env to start hacking at your code ;)