sieren/Homepoint #159: Homepoint and Domoticz?

nahoj58, 2021-02-26:
I'm trying to get data (temperature and humidity) and display thos in Homepoint and maybe also some switches.
But via MQTT Domoticz only publishes domoticz/out. How to select the right device? There is no idx to the gettopic / settopic.
Do I miss something /can somebody help me please.

Johan

htvekov, 2021-03-04:
Hi' @nahoj58

I'm not familiar with Domoticz at all, but googled info about Domoticz MQTT message structure.
Single payload values are apparently not used by Domoticz - only JSON formatted messages. So with current HomePoint version, I don't believe it's possible to control switches or lights as HomePoint is listening for payload values `on` and `off` and not JSON payloads. I believe though, that Matt is currently planning/doing a rewrite that will have the option to check for state in a JSON payload as well.

But you should be able to display a temperature/humidity sensor with somthing like this in HomePoint:
(Assuming temperature data is `svalue1` and humidity is `svalue2` and default Domoticz topic is used)
For this setup you'll also have to load two icons with same name as in `firstIcon` and `secondIcon`

```json
{
    "name": "Greenhouse",
    "type": "Sensor",
    "icon": "garden",
    "devices": [{
      "name": "Greenhouse",
      "type": "combinedValues",
      "firstIcon": "temperature_small",
      "secondIcon": "humidity_small",
      "jsondata": true,
      "firstKey": "svalue1",
      "secondKey": "svalue2",
      "getTopic": "domoticz/out"
    }
```

Ciao !

nahoj58, 2021-03-08:
Hi,

 

I have tried your solution (It looks like the examples on the forum) n[ow  homepoint displays all publisched topics of domoticz . All kinds of values are displayed, depending on how rapid messages with the topic domoticz/out are being published. 

I think that has to do with the way homepoint handles topics: no selection is made by homepoint.

It was nice if homepoint selected the sensor number: “idx”:  xxx is sent by domoticz in the body of the json data/message.

If I could “say” to homepoint to only process a certain ïdx” number I think I would be OK. 

 

Maybe this can be an update or can you program this?

 

Regards,

 

Johan

 

This is an example MQTT message from domoticz:

 

domoticz/out {

        "Battery" : 100,

        "RSSI" : 7,

        "description" : "",

        "dtype" : "Temp + Humidity",

        "hwid" : "2",

        "id" : "50958",

        "idx" : 772,

        "name" : "Woonkamer temperatuur",

        "nvalue" : 0,

        "stype" : "Cresta, TFA TS34C",

        "svalue1" : "20.1",

        "svalue2" : "37",

        "svalue3" : "2",

        "unit" : 5

}

htvekov, 2021-03-08:
Hi' @nahoj58

Sorry, wasn't thinking.
Of course you'll need an identifier, unless you only have one device 🙄

Sorry, this will require a specific Domoticz integration in Home Point. And unfortunately I can't help with the programming part.
When Matt get to the specific Zigbee2MQTT / Home Assistant formats he might include a specific Domoticz mode as well that checks for specified index in JSON message as identifier. The Domoticz format looks pretty straight forward to me.

Alternatively you could convert the Domoticz MQTT messages with MQTTfx, Nodered or similar to act as 'translator' between Domoticz <-> Home Point.

Ciao !

ghosty-be, 2021-03-08:
You need to reconfigure domoticz to publish all messages on an index topic (or you could use name I guess but that would be complexer I guess)
go to settings -> hardware -> click on your MQTT Client Gateway with LAN interface
Then under Publish Topic: pick from the dropdown "Index"
The explanation says: Index - publish outgoing messagen on topic {domoticz/out}/$idx

So for example I have a temperature and humidity sensor in my bedroom which has index 36 in domoticz
I show temperature (in my sensor json output of  svalue1) and humidity (in my sensor json output of svalue2) in 1 scene like this on the homepoint config:

  {
    "name": "Bedroom",
    "type": "Sensor",
    "devices": [{
      "name": "TH",
      "type": "combinedValues",
      "jsondata": true,
      "firstIcon": "temperature_small",
      "secondIcon": "humidity_small",
      "firstKey": "svalue1",
      "secondKey": "svalue2",
      "getTopic": "domoticz/out/36"
    }]
  }
  
Note: the devices name ("TH" in my case) is not important as it's nowhere shown, it only shows the scene name (in my case "Bedroom") 
perhaps that name can even be skipped @sieren? (I just based this snippet on examples I found...)

Here is another example where I combine 2 different (temperature) sensors in 1 scene:
{
    "name": "AC Sensor",
    "type": "Sensor",
    "devices": [{
      "name": "t_inside",
      "type": "singleValue",
      "jsondata": true,
      "firstIcon": "temperature_small",
      "firstKey": "svalue1",
      "getTopic": "domoticz/out/74"
    },
    {
      "name": "t_outside",
      "type": "singleValue",
      "jsondata": true,
      "firstIcon": "temperature_small",
      "firstKey": "svalue1",
      "getTopic": "domoticz/out/73"
    }]
  },

nahoj58, 2021-03-09:
Hi,

 

There is under the publish topic no item “index” I can only pick “out”, “/” or both

 

The domooticz version is 2020.2 build hash b63341bc0 and the compile  date 04-26-2020.

 

Checking for updatessays I have the latest stable version

 

????

htvekov, 2021-03-09:
Solution: Shift to beta 👍😆

nahoj58, 2021-03-10:
> oh I run betas ... :)
>  2020.2 build 13058 (dated 2021-03-08)

No problemen yet, I'm a bit careful runnning
  
Beta's ?

nahoj58, 2021-03-10:
Can I do Thatcher without problems?

htvekov, 2021-03-12:
Hi' @nahoj58 

I'm sorry, but I have absolutely no idea about the stability of Domoticz betas.
Don't even know the program at all.

I would suggest you googled for issues and made a full backup of Domiticz, data and setup before trying it out.

ghosty-be, 2021-03-15:
I run it already a couple of years in beta and despite a couple hiccups here and there (like some graphs not fully plotting or half invisible (you could see the values if you mouseover, but the actual lines were missing) 
I have not really had any serious issues... 
But as @htvekov said: backup and test is the best way forward :)
Also the beta gets like updates every couple of days ...

nahoj58, 2021-03-15:
Thanks,

 

What is the best way to back-up or how to go back to a stable release or older beta?

 

Just make a copy of your SB-card?

 

 

An other question ( or is tere a forum item?):

 

My domoticz (with a 433 antenne and a zwave-stick) runs on a pi 2B+, I like to migrate to a pi 4B

How to do that the easy(iest) way?

ghosty-be, 2021-03-16:
my domoticz is just installed in /home/pi/domoticz (I guess thats according to the standards... ? )  
since you just run in that directory the "updatebeta" script ... 
just keeping a copy around of the whole directory contents is enough I suppose ... :)

nahoj58, 2021-03-16:
Can you tell me how to make a copy of this directory to your HD and copy this directory back?

 

I ‘m just a windows user.

ghosty-be, 2021-03-20:
you need to install something like winscp ... if you configure that to connect to your raspberry pi you can just browse through it and download stuff...
But anyway this is beyond the scope of the original question... for domoticz support see the domoticz pages I'd say...

nahoj58, 2021-04-01:
I have installed 2020.2 build 13133 (dated 2021-03-30)

 

But I can choose in the menu “Publish Topic” the same options as in the stable version 2020.2 (dated  2020-04-26 ), there is  no option “index” to choose.

 

Have I donesomething wrong: I installed the stable Domoticz and there after the beta (./updatebeta in the domotcz directory)

Do  have to install something extra?