sieren/Homepoint #163: Laggy experience with freezes out of the box m5core2

jgilbert20, 2021-03-08:
I feel like I must be doing something wrong, but wanted to log this in case someone else is experiencing it too.

Running a very simple config, I'm finding a very simple switch is not triggering immediately. Instead, the button presses don't seem to activate anything, or are very lagged and occasionally just call full resets. 

Flashed the firmware with this command:

esptool.py -b 921600 write_flash -fs 4MB -fm dout 0x0 homepoint_m5stackcore2_full.bin

I have other MQTT connected devices running (including homebridge) and they seem to be running very fast.

Configuration:

```
{
    "wifi": "XXX",
    "password": "XXX",
    "login": "admin",
    "webpass": "XXX",
    "mqttbroker": "mqtt://10.0.0.48",
    
     "scenes": [{
    "name": "Living Room",
    "type": "Light",
    "icon": "kitchen",
    "devices": [{
      "name": "Side Light",
      "setTopic": "GoodMorningSwitch_set",
      "getTopic": "GoodMorningSwitch_set",
      "onValue": "true",
      "offValue": "false",
      "icon": "bedroom"
    }]
  }]
}
```

I generally find that the icon is flipping properly with I nudge the topic GoodMorningSwitch_set from the command line, and that is pretty quick. But pressing the button often causes pauses or even full crashes and reboots.

In this log file, I was pressing the living room button repeatedly. It seemed to respond to about 30% of my presses. And then there was a full crash. 

Appreciate any thoughts. 

`14:11:50.220 -> 
14:11:50.220 ->      .0 .1 .2 .3 .4 .5 .6 .7 .8 .9 .A .B .C .D .E .F 
14:11:50.220 -> 
14:11:50.220 -> 0.   00 00 00 40 49 00 5F 00 00 FF FF FF FF FF FF FF 
14:11:50.220 -> 1.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> 2.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> 3.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> 4.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> 5.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> 6.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> 7.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> 8.   26 00 0F 00 00 C8 01 1E 0A 28 00 00 00 00 00 00 
14:11:50.220 -> 9.   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 26 
14:11:50.220 -> A.   02 30 0E 64 01 00 10 00 11 0F 00 00 00 00 00 01 
14:11:50.220 -> B.   00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 
14:11:50.220 -> C.   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
14:11:50.220 -> D.   00 FF FF 00 00 FF FF FF FF FF 00 00 FF FF FF FF 
14:11:50.220 -> E.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:50.220 -> F.   FF FF FF FF FF FF FF FF FF FF FF FF 01 FF FF FF 
14:11:50.220 -> 
14:11:50.220 -> 
14:11:55.472 -> 
14:11:55.472 ->      .0 .1 .2 .3 .4 .5 .6 .7 .8 .9 .A .B .C .D .E .F 
14:11:55.472 -> 
14:11:55.472 -> 0.   00 00 01 80 53 00 59 00 00 FF FF FF FF FF FF FF 
14:11:55.472 -> 1.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:55.472 -> 2.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14:11:55.472 -> 3.   FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
[... 1082 lines cut ...]
14:13:05.967 -> I (2217) wifi:state: assoc -> run (10)
14:13:06.004 -> I (2223) wifi:state: run -> init (3c0)
14:13:06.004 -> I (2223) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
14:13:06.004 -> I (2223) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
14:13:06.004 -> [0;31mE (2229) event: system event loop not initialized via esp_event_loop_init[0m
14:13:06.004 -> [0;32mI (2237) wifi station: retry to connect to the AP[0m
14:13:08.036 -> [0;31mE (4284) event: system event loop not initialized via esp_event_loop_init[0m
14:13:08.036 -> [0;32mI (4286) wifi station: retry to connect to the AP[0m
14:13:08.179 -> I (4406) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
14:13:08.179 -> I (4406) wifi:state: init -> auth (b0)
14:13:08.179 -> I (4408) wifi:state: auth -> assoc (0)
14:13:08.179 -> I (4411) wifi:state: assoc -> run (10)
14:13:08.179 -> I (4427) wifi:connected with Sapphire, aid = 1, channel 11, BW20, bssid = 48:4b:d4:58:d7:d1
14:13:08.179 -> I (4428) wifi:security: WPA2-PSK, phy: bgn, rssi: -37
14:13:08.179 -> I (4432) wifi:pm start, type: 1
14:13:08.179 -> 
14:13:08.179 -> [0;31mE (4433) event: system event loop not initialized via esp_event_loop_init[0m
14:13:08.254 -> I (4515) wifi:AP's beacon interval = 102400 us, DTIM period = 1
14:13:09.231 -> [0;31mE (5456) event: system event loop not initialized via esp_event_loop_init[0m
14:13:09.231 -> [0;32mI (5456) tcpip_adapter: sta ip: 10.0.0.50, mask: 255.255.255.0, gw: 10.0.0.1[0m
14:13:09.231 -> [0;32mI (5460) wifi station: got ip:10.0.0.50[0m
14:13:09.231 -> [0;32mI (5465) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
14:13:09.268 -> [0;32mI (5506) MQTT: MQTT_EVENT_CONNECTED[0m
14:13:09.268 -> [0;32mI (5516) MQTT: MQTT_EVENT_SUBSCRIBED, msg_id=37814[0m
`

sieren, 2021-03-08:
Sadly this is currently a known issue... tracking here https://github.com/sieren/Homepoint/issues/145

closed 2021-03-08 by sieren

Girgitt, 2024-01-23:
The problem with i2c-related crashes is due to rx buffer overflow in underlying arduino-esp32. It is fixed in PR https://github.com/espressif/arduino-esp32/pull/4342 but unfortunaltelly it is not that easy to use relrease version of arduino-esp32 with this fix as it requires at least v2.0.2 which requires esp-idf v4.4-beta1 which breaks Homepoint (and with changes to make Homepoint compile with arduino-esp32 v2.0.2 or v2.0.3 the initial AP mode with default empty wifi settings is unstable). 

But anyway if anyone is interested I made fixes to the Homepoint v1 (non lvgl-based) version in my fork: https://github.com/Girgitt/Homepoint/tree/v1-m5core2-fixes 
In specific I back-ported PR 4342 (and had to switch upstream repo to my for of sieren's fork of arduino-esp32) 
There are also two other minor fixes like corrected color translation from 24b to 16b and disabled screensaver which does not deactivate.