sieren/Homepoint #167: AZ Touch Mod   2,8" Display

detafun, 2021-03-29:
Hello,
I have the AZ-Touch Mod with the 28" Display. Installed on the ESP32 with the current release and installed the File
"homepoint_aztouchmod_full.bin "there. So far so good. But unfortunately the display does not react correctly when you press the buttons the display.

To test whether my display is ok, I installed the "https://www.hwhardsoft.de/app/download/11665157197/CodeLock_ESP.zip?t=1604908389" It works perfectly, so my display is ok. What I found when I set my test installation to 2.4", then it does the same nonsense as with Homepoint. If I set my test installation to 2.8", it works perfectly. I suspect that the 2.8" Display does not work properly with Homepoint. And it treats it as if it were a 2.4" Display.

What should I do?

cu

llevet, 2021-04-04:
Hi @detafun 
I have the AZ-Touch Mod with the 2.8" Display too.
I had the same problem, 

It work now for me : 
Try to use v0.07.2 and update with homepoint_espgeneric and not homepoint_aztouchmod file.
I also change the config and add : 
"touchXAxisInverted": true,
"touchYAxisInverted": false,

I don't explain the problem, I electrically verified the PCB and the T_IRQ is well connected to GPIO 27.
I have compile by myself the actual GIT version and only homepoint_espgeneric_ota_update.bin is working (not the version homepoint_aztouchmod_ota_update.bin).

After compilation i verify the file : /root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup.h and it containt : #define TOUCH_IRQ 2 and it work ! 

If i make : cmake -DBOARD=AZTOUCHMOD ../
the file /root/esp/homepoint/main/libraries/TFT_eSPI/User_Setup.h containt : #define TOUCH_IRQ 27 and it doesn't work !

It is so weird  ! 
@sieren if you have a explanation i take it !

franki29, 2021-04-05:
Hi ,
I am using same device but works OK with AZ-Touch MOD Enclosure firmware using:
screenRotationAngle": 1, in the config.json file.
Br. Frank

detafun, 2021-04-05:
Hello llevet, 
I have now installed homepoint_espgeneric v0.07.2 full.bin. 
Then in the config the two lines
"touchXAxisInverted": true,
"touchYAxisInverted": false,
inserted, but unfortunately it still doesn't work. 

Can you maybe upload with your bin? Would be great until @sieren fixed it.

detafun, 2021-04-05:
> Hi ,
> I am using same device but works OK with AZ-Touch MOD Enclosure firmware using:
> screenRotationAngle": 1, in the config.json file.
> Br. Frank

Mh, in the Zip is not a AZ-Touch MOD Enclosure file.
```
homepoint_aztouchmod_full.bin        homepoint_espgeneric_full.bin        homepoint_m5stackcore2_full.bin        homepoint_m5stack_full.bin        readme.txt
homepoint_aztouchmod_ota_update.bin  homepoint_espgeneric_ota_update.bin  homepoint_m5stackcore2_ota_update.bin  homepoint_m5stack_ota_update.bin

```

llevet, 2021-04-05:
Hi,
homepoint_aztouchmod_full.bin and homepoint_aztouchmod_ota_update.bin is for AZ-Touch MOD in zip.

Try this file, it is the working file for me (build from github source version of 04/04/2021). The update must be OTA.
[homepoint_espgeneric_ota_update.zip](https://github.com/sieren/Homepoint/files/6258675/homepoint_espgeneric_ota_update.zip)

llevet, 2021-04-05:
Hi  @detafun 

In fact when you said : 
But unfortunately the display does not react correctly when you press the buttons the display

Is the display react ? Could you hear the beep when you click ?  Clicking does something ? 
Or is it simply that the area you click on is not the same as the one that reacts ?

franki29, 2021-04-05:
I have used homepoint_aztouchmod_full.bin  , sorry for being unclear.
And I had to use screenRotationAngle": 1 otherwise the click on a Icon was  not working.

llevet, 2021-04-05:
Ok,
This is my full version working on my AZ-Touch Mod with the 2.8" Display.

[homepoint_espgeneric_full.zip](https://github.com/sieren/Homepoint/files/6259373/homepoint_espgeneric_full.zip)

detafun, 2021-04-05:
> I have used homepoint_aztouchmod_full.bin , sorry for being unclear.
> And I had to use screenRotationAngle": 1 otherwise the click on a Icon was not working.

Yes, that's how it works. Thank you

llevet, 2021-04-05:
Glad to hear your problem is resolved. 

"touchXAxisInverted": true,"touchYAxisInverted": false, as the same result as "screenRotationAngle": 1 , this 2 methods are used to put in sync screen and touchpad.
The first is rotate the touchpad, the second is rotate the screen.

For my part, I prefer to rotate the touchpad, because the power supply part of the PCB is on top, and the heat stays on top, and didn't move from the bottom to the top crossing all the pcb.

So i  just testing homepoint_aztouchmod_full.bin and it is working now like homepoint_espgeneric_full.bin too ...
This mean the code is not using only interrupt for touchpad but use also pooling if it work with the 2 versions ...
I don't know why the first time it doesn't ...

sieren, 2021-04-05:
Hey folks, thanks for providing support. I need to fix the defaults in one of the next updates. Realized recently it's not defaulting to the right values for this platform.

sb185, 2021-04-10:
> Ok,
> This is my full version working on my AZ-Touch Mod with the 2.8" Display.
> 
> [homepoint_espgeneric_full.zip](https://github.com/sieren/Homepoint/files/6259446/homepoint_espgeneric_full.zip)

Hi,
unfortunately i have the same Problem with the AZ Touch Mod 2,8" Display.

I try use the .bin from @llevet to fix it.
But it doesn't work.

Now i have the "Click" Sound when im tapping on the Display, but the
Display is also mirrored. :(

[See Video for Detail Info](https://youtu.be/BULyoPjjgu0)

Otherwise great work @sieren! :)

llevet, 2021-04-10:
Hi @sb185 

Put It in your config file  and restart : 
"touchXAxisInverted": true,
"touchYAxisInverted": false,

sb185, 2021-04-10:
Thank you @llevet. 

It works great! :+1: 

For others... I append the two lines in config.json.:

[image]

sieren, 2021-04-10:
Thanks for providing these. Next release will do without manually configuring the display/touch rotations.

closed 2021-04-10 by sieren