sieren/Homepoint #171: sonoff mqtt adapter iobroker

detafun, 2021-04-06:
Hello, 
new problem what I just discovered. 

I have iobroker sonoff mqtt adapter running. 
The point to switch is correctly set from true to false and back. 
But nothing happens on the device. It doesn't turn on or off. 
If I switch the point manually myself, then it works perfectly, just like when I switch the point via the normal visualization. Just not from the home point. 

Anyone have an idea?

      "name": "Wandlampe",
      "setTopic": "sonoff/0/Wandlampe/POWER",
      "getTopic": "sonoff/0/Wandlampe/POWER",
      "onValue": "true",
      "offValue": "false"

franki29, 2021-04-07:
Hi, are you sure that" true" and "false" are the right values?
Did you flash a alternative software into your sonoff device like tasmota?
Maybe the easiest way to find out the right commands you should use is to listen to your MQQT broker and look what values are used if you switch your devices ion and off.
I am using http://mqtt-explorer.com/ on Linux. 
I am using tasmota with sonoff and the values are 1 for on and 0 for off.

Br. Frank

detafun, 2021-04-07:
Problem has been solved. there has to be a /set at the end.
 "setTopic": "sonoff/0/Wandlampe/POWER/set",

cu Deta

closed 2021-04-07 by detafun