sieren/Homepoint #173: HomePoint

nahoj58, 2021-04-10:
I have two questions:

I have installed homepoint on a M5STACK to display sensors out of Domoticz.

How do I change to time the display stays on, or stays on for ever? Now the display turns to black and none of the keys below the screen function to turn the dspay on again.

How do I add exta icons to the program to choose from like an icon for air pressure, electrriity or gsas for readings of a smart meter. I discovered the data folder withe the icons displayed in the homepoint program but how to add extra ones ?

Johan

ghosty-be, 2021-04-15:
thats odd... mine wakes up from keypress on one of the keys of my m5stack ... the left and middle are to move left or right in the tiles and the right button is used to "select" a tile (if it's a button to toggle it on or off).
I have however attached the m5stack to constant USB power because the builtin battery looked like crap... (had to manually recharge it from a lab power supply because it didn't want to hold a charge through USB... 
you could also extend the on-time with:
  "screenSaverMinutes": 60, 
  (that should keep it on for 1h)

nahoj58, 2021-04-19:
How do I add (new/other) icons tot he webinterface of homepoint?