sieren/Homepoint #174: Screen randomly coming on

cerietke, 2021-04-17:
I've placed an M5stack Core on my bedside table and use it to turn off the light and other stuff in the bedroom when I go to bed. Every now and then I notice that the screen comes on in the middle of the night (mostly when it's shining right in my face and wakes me). I don't believe anything is being sent to it on MQTT. Is this standard behavior for Homepoint or is it something else? Is there a configuration option to make it stop doing this? If this is not Homepoint, anybody an idea how I can best figure out what it's responding to?

Possibly an enhancement idea: being able to set operating hours outside of which the screen only comes on based on a button press?

ghosty-be, 2021-05-09:
isn't it rebooting at that point? (see other bugs where we reported random reboots when it looses wifi or something)
I have never seen any other reason why it would light up (at least in my testing)

cerietke, 2021-05-10:
It's possible: as it comes from a black screen it's hard to tell what happens before. There are no indications of issues that I've seen. I can control my light through it and it looks the same as when I turn my light on through google assistant instead, which through my automation updates the homepoint state for that switch.